bambulab/BambuStudio #2876: Mesh Boolean: result jumps to new coordinate

coat64, 2023-11-12:
### Bambu Studio Version

1.8.0.57

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

CPU: AMD 5700G
Mem: 64GB

### Printer

X1C

### How to reproduce

Start Mesh Boolean
Choose Union (or difference)
Select two parts in assembly
Press Union

### Actual results

The new object/part jumps to a new coordinate, very annoying if other parts are involved.
Before union:
[image]
After Union:
[image]

### Expected results

The result stays on same position.

### Project file & Debug log uploads

[testunion.zip](https://github.com/bambulab/BambuStudio/files/13328263/testunion.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SaltWei, 2024-01-17:
检查一下是否和坐标系问题是一类根因

jslamartina, 2024-03-14:
I noticed this too. If I want to union a complex set of parts, they jump around and I lose the precision of my original assembly, which is a pretty serious issue when dimensions need to be exact.