bambulab/BambuStudio #2878: Correct the strings

AndylgTom, 2023-11-12:
### Bambu Studio Version

Bambu Studio_v1.8.0 Beta

### Where is the application from?

Bambu Lab github releases

### OS version

win10

### Printer

never mind

### How to reproduce

Please correct the strings in the screenshots. 
On the first screenshot - The string occurs in two places in different values. It would be more logical to name it "Share" at the top and write "Load" in the table model loading.
On the second screenshot, correct the writing of languages.
Thank you.
[image]
[image]

### Actual results

аааааааа

### Expected results

ааааааааааа

### Project file & Debug log uploads

ааааааааа

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DanBao-Bambu, 2023-11-14:
May I ask if the spelling of Ukrainian is accurate? The spelling displayed on Google is "Українська".

AndylgTom, 2023-11-14:
> Українська

Yes, Украї**н**ська is correct

closed 2024-01-18 by DanBao-Bambu