bambulab/BambuStudio #2880: Crashing while inserting Benchy Primitiv

Roughhawk, 2023-11-13:
### Bambu Studio Version

Beta 1.8.0.57

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10 Enterprise

### Additional system information

AMD Ryzen 5 5625U with Radeon Graphics            2.30 GHz
16 GB RAM

### Printer

Bambu P1S

### How to reproduce

Prepare
Rightklick at Buildplate
'Add Primitiv'
SW closing without message

### Actual results

SW closing without message

### Expected results

3D Benchy shall be placed on buildplate

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13331980/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Roughhawk, 2023-11-13:
[image]

DanBao-Bambu, 2023-11-14:
There were some errors while reading the file. Do you still have this problem now? If so, you can try installing Studio to a different directory.

closed 2023-11-14 by Roughhawk