bambulab/BambuStudio #2881: Selecting Generic ASA shows PC on AMS ext spool

CdRsKuLL, 2023-11-13:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

X1 Combo (AMS)

### How to reproduce

Go on device, select Generic ASA from the Ext Spool and it shows PC but if you select Bambu ASA, it shows ASA

### Actual results

Shows PC not ASA

### Expected results

To show ASA

### Project file & Debug log uploads

Nothing to upload

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-11-23:
Hi. @CdRsKuLL 
Please use new version.
https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.62

walterwongbbl, 2023-12-21:
We have applied the latest version, please download the latest version to experience it.

closed 2023-12-21 by walterwongbbl