bambulab/BambuStudio #2883: SVG Import error.  Fill is in the wrong spot

onetgarret, 2023-11-13:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 14.2 (23C5030f)

### Additional system information

CPU: M1 Pro

### Printer

P1S

### How to reproduce

1. import SVG (peace.svg)
2. notice incorrect logo

### Actual results

image doesn't have the fill in the correct place
[image]

### Expected results

fill in the correct place
[image]

### Project file & Debug log uploads

SVG file to import
[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2023-11-14:
@onetgarret  Could you share your svg file?

onetgarret, 2023-11-14:
This is the file: https://user-images.githubusercontent.com/299274/282524127-7b5f2851-24b4-4638-a34a-82b054ed7c3e.svg

zhimin-zeng-bambulab, 2023-11-16:
[image]

zhimin-zeng-bambulab, 2023-11-16:
@onetgarret  Sorry, I can't download this svg file here. If you can't upload it directly, you can compress it into a zip file and upload it.

zhimin-zeng-bambulab, 2023-11-16:
Thanks, I have got it

Haidiye00, 2024-10-16:
[image]
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00