bambulab/BambuStudio #2885: Tree Organic Support has issues.

mhwunder, 2023-11-14:
### Bambu Studio Version

1.8.0.57

### Where is the application from?

Bambu Lab github releases

### OS version

Mac 14.1

### Printer

P1P

### How to reproduce

Slice using Organic Tree support.  Works properly with Tree Slim

### Actual results

Support Settings:
[image]

Before slice:
[image]

After:
[image]

### Expected results

Proper support structure.

### Project file & Debug log uploads

My file is too large to upload.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tangketan, 2023-11-15:
Is there any chance that you downsize the file and upload it? Without the 3mf file I can't do anything...

mhwunder, 2023-11-15:
Can I send you a link?

Regards,

Michael Wunder

Direct: (510) 984-1660

Mobile: (925) 817-0119

Email: ***@***.***

Quote: ***@***.***

mhwunder, 2023-11-16:
Attached is a reduced file.  Some additional info.

If I just slice the model upon import after scalling and positioining, the Organic Tree works.  After I change the Adaptive Variable Layer Height all the way to the left and click the Adaptive button and then slice.  The slicing gets all messed up.

[image]

[Odd Organic Support 2.3mf.zip](https://github.com/bambulab/BambuStudio/files/13379807/Odd.Organic.Support.2.3mf.zip)

tangketan, 2023-11-23:
Organic support doesn't work well with adaptive layer height. I'll look into it.

seva-luchianov, 2023-12-15:
+1 seeing issues with adaptive layer height and organic tree support. Seems like the tree support stays at the same layer height throughout and falls behind from the height of the model
[image]

tangketan, 2023-12-21:
@seva-luchianov Yes, you are right. For now please use other support styles.