bambulab/BambuStudio #2889: Bambu Studio CLI cannot load settings from JSON

eropple, 2023-11-14:
### Bambu Studio Version

v01.08.00.57

### Where is the application from?

Bambu Lab github releases

### OS version

Fedora 39

### Additional system information

n/a

### Printer

n/a

### How to reproduce

When you look at `/path/to/BambuStudio.AppImage --help`, you'll see:

```
                     Load filament ids for each object
 --load-filaments "filament1.json;filament2.json;..."
                     Load filament settings from the specified file list
 --load-settings "setting1.json;setting2.json"
                     Load process/machine settings from the specified file
```

This implies two things:
- both `--load-filaments` and `--load-settings` can take lists of files containing the appropriate settings for each.
- both support JSON files, presumably the files exported from Bambu Studio.

Neither are true. To reproduce:

- Export from Bambu Studio the relevant profile files.
- Attempt to export a STL that uses `--load-filaments and/or `--load-settings` in a reasonable way. For example, try something like the following:

```
$ ~/Downloads/Bambu_Studio_linux_fedora_v01.08.00.57-20231109141031.AppImage \
    --avoid-extrusion-cali-region \
    --outputdir ./tmp/output \
    --export-3mf \
    --load-settings "ed3dnet_-_Bambu_Lab_P1S_0.4_nozzle.json;ed3dnet_-_0.20mm_Standard_@BBL_X1C.json" \
    --load-filaments "ed3dnet_PLA.json;ed3dnet_PLA_Matte.json;.ed3dnet_PETG.json" \
    connector.stl \
    box.stl \
    drawer.stl \
    nameplate.stl
[2023-11-14 17:11:41.618313] [0x00007f6ad15ced80] [trace]   Initializing StaticPrintConfigs
No such file: ed3dnet_-_Bambu_Lab_P1S_0.4_nozzle.json;ed3dnet_-_0.20mm_Standard_@BBL_X1C.json
run found error, return -3, exit...
```

- Since that doesn't work (because this clearly isn't parsed correctly), we reduce to a single setting/filament as per the following:
```
$ ~/Downloads/Bambu_Studio_linux_fedora_v01.08.00.57-20231109141031.AppImage \
    --avoid-extrusion-cali-region \
    --outputdir ./tmp/output \
    --export-3mf \
    --load-settings "ed3dnet_-_Bambu_Lab_P1S_0.4_nozzle.json" \
    --load-filaments "ed3dnet_PLA.json" \
    connector.stl \
    box.stl \
    drawer.stl \
    nameplate.stl
[2023-11-14 17:11:54.747784] [0x00007efc6feefd80] [trace]   Initializing StaticPrintConfigs
ed3dnet_-_Bambu_Lab_P1S_0.4_nozzle.json: Unknown file format. Input file must have .stl, .obj, .amf(.xml) extension.
```

### Actual results

See above.

### Expected results

I would expect Bambu Studio to load the relevant filament, printer, and process settings, then do something reasonable with them in order to generate a 3mf that includes those configuration values.

### Project file & Debug log uploads

The CLI does not emit logs.

There is no project file to upload; that's the point of running this command.

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2023-11-15:
you need to add a file_name behind --export-3mf

~/Downloads/Bambu_Studio_linux_fedora_v01.08.00.57-20231109141031.AppImage \
    --avoid-extrusion-cali-region \
    --outputdir ./tmp/output \
    **--export-3mf  output.3mf** \
    --load-settings "ed3dnet_-_Bambu_Lab_P1S_0.4_nozzle.json" \
    --load-filaments "ed3dnet_PLA.json" \
    connector.stl \
    box.stl \
    drawer.stl \
    nameplate.stl

eropple, 2023-11-16:
Sorry - I copied that poorly, there is an argument there. This leads to other errors and a lack of clarity, though - specifically that what is exported _from_ Bambu Studio_ can't in turn be used to run from the command line. In the above cases, those exported filament, printer, and process JSON files don't work; you need a "canonical" one, it seems, such as the BambuSlicer profiles under `resources/profiles`. There's something odd to the parsing there. It's not that it can't load any, but it can't load what you'd expect.

(that `--export-3mf` does not mirror `--export-stl` is very strange, though.)

lanewei120, 2023-11-16:
please refer to https://github.com/bambulab/BambuStudio/issues/1963
for more details discussed before

you need to generate the full jsons including the inherit values from their inherited jsons

eropple, 2023-11-16:
Got it. That's really confusing, but it unblocks me. Thanks.

closed 2023-11-16 by eropple