bambulab/BambuStudio #2892: AMS reads a PLA RFID tag correctly, but Bambu studio shows PETG-CF

JeffChapman, 2023-11-15:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro

### Printer

Bambu X1C

### How to reproduce

1. Load orange PLA-basic in slot 4 (Bought through Bambu)
2. AMS reads it as PLA (color orange) 
[image]
[image]
3. Resync AMS in Bambu Studio
5. It correctly reads the filament as 'Bambu PLA Basic'
[image]
6. Go to print, and I can't select slot 4 as it doesn't match the sliced material
[image]

Notes: 
1. I have also tried debugging by trying slot 1/2/3, as well as loading a Support-PLA that came with the printer in hopes that it would refresh things somehow
[image]
[image]
[image]

2. I have a black PETF-CF loaded in slot 1, but removing it and tried the PLA-Basic in that slot as well to no avail 

### Actual results

Details above as well, but Bambu studio reads PLA-Basic RFID as PETG-CF.

### Expected results

Details above as well, but Bambu studio should be reading PLA-Basic as PLA-Basic so that I can select it when trying to print a PLA sliced model.

### Project file & Debug log uploads

[trays x5 colors project.zip](https://github.com/bambulab/BambuStudio/files/13367574/trays.x5.colors.zip)
[debug_Wed_Nov_15_11_57_18_9180.log.zip](https://github.com/bambulab/BambuStudio/files/13367584/debug_Wed_Nov_15_11_57_18_9180.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2023-11-23:
Hi. @JeffChapman 
Please use new version.
https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.62

JeffChapman, 2023-11-24:
Latest version of BambuStudio fixes the issue, thanks!

closed 2023-11-24 by JeffChapman