bambulab/BambuStudio #2893: Bambu Studio shows ASA instead of ABS in AMS when attempting to print

Kiwistylebuild, 2023-11-15:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1-C

### How to reproduce

Slot 1 in AMS selected as Generic ABS. 
[image]

Bambu Studio Resunc with AMS. Part selected as slot 1 Generic ABS.  
[image]

Slice Plate. 
Print Pate.
[image]

Shows unable to print due to slot 1 being ASA, not ABS

### Actual results

[image]

Shows unable to print due to slot 1 being ASA, not ABS

### Expected results

Show ABS in slot 1

### Project file & Debug log uploads

[cap.zip](https://github.com/bambulab/BambuStudio/files/13367902/cap.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/13367915/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Kiwistylebuild, 2023-11-21:
I just setup a brand new x1c with two AMS and the same issue has happened.  It has labelled them all as ASA when trying to print.
[image]

walterwongbbl, 2023-11-23:
Hi.
Please use new version.
https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.62

walterwongbbl, 2023-12-21:
We have applied the latest version, please download the latest version to experience it.

closed 2023-12-21 by walterwongbbl