bambulab/BambuStudio #2894: Anomalies originating (probably) from 'Fuzzy' modifier and file type variations (.step and .stl)

manuelmvsosa, 2023-11-15:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Nombre del SO	Microsoft Windows 10 Home Single Language
Versión	10.0.19045 compilación 19045
Fabricante del SO	Microsoft Corporation
Fabricante del sistema	LENOVO
Modelo del sistema	81LK
Tipo de sistema	PC basado en x64
SKU del sistema	LENOVO_MT_81LK_BU_idea_FM_L340-15IRH
Procesador	Intel(R) Core(TM) i5-9300H CPU @ 2.40GHz, 2400 Mhz, 4 procesadores principales, 8 procesadores lógicos
Versión y fecha de BIOS	LENOVO BGCN33WW, 24/11/2020
Versión de SMBIOS	3.0
Versión de controladora integrada	1.33
Modo de BIOS	UEFI
Fabricante de la placa base	LENOVO
Versión de la placa base	SDK0K17763 WIN
Memoria física instalada (RAM)	16,0 GB
Memoria física total	15,9 GB
Memoria física disponible	5,25 GB
Memoria virtual total	32,3 GB
Memoria virtual disponible	16,9 GB
Espacio de archivo de paginación	16,4 GB
Display card: NVIDIA GeForce GTX 1650

### Printer

Bambu Lab X1 Carbon Combo

### How to reproduce

1. Open any .3mf file provided. All of them were created using Bambu Studio. 
2. For files 01, 01-1 and 03, slice any model and check for the anomalies described in the "Actual results" section. In file 02 you won't find any evident issues. 
3. Print any plate and hopefully you will get the same result as me. 

### Actual results

To date, I've encountered challenges with several components, and, in fact, some have only been successfully printed under very specific conditions. After a thorough investigation of the issue and extensive online research, my conclusion is that it is related to Bambu Studio. However, I have yet to find a solution or any reference to address these specific problems.

More specifically, this issue has surfaced in three distinct parts, which I will detail below using the internal names employed within our company:

1. LH_Bio Cabinet: While working from the same 3D model, I observed disparities between the .step file (left) and .stl file (right). In the case of the .stl file, an unusual extrusion line with a considerable diameter appears in the demarcated area. It seems this situation arises from the 'fuzzy' modifier. Additionally, adding more than one piece to the print bed causes issues (likely over-extrusion), resulting in threads and small balls that damage the printed part.

[image]
[image]
[image]

2. Electronics Box Lid for Dryer: Similar to the preceding issue, complications arise (presumably from over-extrusion), creating threads and small balls that damage the printed part. Interestingly, the use of the fuzzy modifier in this case also seems to contribute to the problem.

4. MC-200 Cabinet: Using the .step file leads to over-extrusion problems similar to those described above. As for the .stl files, printing 1 or 2 pieces works well, but attempting to print 4 generates a problem, as depicted in the attached photo—manifesting as extremely large balls from the 2nd layer onward.

[image]
[image]
[image]

In all instances, the large and small balls become obstacles that the hotend encounters, sometimes with significant force. In summary, the primary culprit for these failures could be the 'fuzzy' command, and using .step or .stl files, even from the same source, can yield different results. Additionally, in all cases, I was unable to identify any discernible pattern leading to the failure or the distinct behaviors.

I appreciate your attention to this matter. Feel free to reach out if further information is needed, and I apologize if I did not explain any section clearly. I am at your disposal to provide additional clarification or details. Thank you for your understanding.

### Expected results

Perfect prints as happens with the other files I have printed without issues with the sliced parts.  

### Project file & Debug log uploads

[bambu_studio_gitHub_issue.zip](https://github.com/bambulab/BambuStudio/files/13369269/bambu_studio_gitHub_issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2023-11-24:
@sodachen22 please check issue 3. and i will check issue 1 and 2

QingZhangBambu, 2023-11-24:
@manuelmvsosa 
**for issue 1:** 
i change the color of the modifier part to explain the reason, this part is too small to generate the wall , the contour didn't generate. so the fuzziy skin dose not work. 
[image]
if you move the modifier a bit inside , the outwall will be fine.
[image]

**for issue 2:**
1.8.0 beta 2 works fine, you can use that version.
[image]

closed 2025-05-30 by QingZhangBambu