bambulab/BambuStudio #2895: Trying to run a business here. Slicer not working with printers like they should.

TheIndoCloud, 2023-11-15:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

P1S

### How to reproduce

after slicing and sending to the servers, the device screen appears and printer and slicer are both stuck at downloading 0%

### Actual results

both device and printer frozen and slicer keeps closing on its own repeatedly when I try to communicate with printer.
This has been happening since I got my new addition last week.

### Expected results

it should download to printer and start printing the thing.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13369584/log.zip)
[5.0_telescope_wand_one_piece.zip](https://github.com/bambulab/BambuStudio/files/13369587/5.0_telescope_wand_one_piece.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

TheIndoCloud, 2023-11-15:
I changed the wifi on the printer which seems to have solved the download issue but that does not fix my issue with the slicer automatically closing during some projects. It does not always restore what I had on slicer.

walterwongbbl, 2023-11-23:
@sodachen22 Please check the printer's log

walterwongbbl, 2023-11-23:
Hi. @TheIndoCloud 
Can you record a video or take a photo? I want to see how you operate.

closed 2024-01-18 by walterwongbbl