bambulab/BambuStudio #2896: Slicer should warn if installed nozzle size mismatch detected (between slicer nozzle size vs. configured nozzle on machine)

MovingSymbols, 2023-11-15:
**No warning if there is nozzle size mismatch between slicer and X1C hardware configuration**

I forgot I had a 0.2 mm nozzle on my X1C and sent a job sliced for a 0.6 mm nozzle. The job started with no warnings or errors but did not complete properly.

Of course the X1C and Bambu Slicer of course have no way to detect the actual nozzle size, _but in this case_ Bambu Slicer should have created a warning when attempting to send the job to the printer, because a mismatch in configuration existed.

**Proposed Solution**

If the X1C is configured (via the touchscreen) as having a nozzle size installed that does not match the slicer's settings when a print job is sent to the printer, Bambu Slicer should create a warning.

**Steps to recreate issue**

1. X1C configured with a Stainless Steel 0.2mm nozzle (done via touchscreen in settings on the printer)
2. Open and slice a job in Bambu Slicer configured for a 0.6 nozzle.
3. Send print job to the printer, no errors or warnings.
4. Print job does not complete properly due to nozzle size mismatch.

**Expected Behavior**

1. X1C configured with a Stainless Steel 0.2mm nozzle (done via touchscreen in settings on the printer)
2. Open and slice a job in Bambu Slicer configured for a 0.6 nozzle.
3. Send print job to the printer, **Bambu Slicer detects mismatch between print job slicer setting (0.6 nozzle) and what X1C reports as being installed by the user (0.2 nozzle) and warns the user.**
4. User can ignore warning and proceed, or cancel job.

marko-hologram, 2023-11-16:
https://github.com/bambulab/BambuStudio/releases

Check the latest 1.8.0 public beta. It does mention "Support nozzle diameter detection and nozzle hardness detection". Is this what you are looking for?

MovingSymbols, 2023-11-16:
That does look like the same feature, thanks for the heads up!

closed 2023-11-16 by MovingSymbols