bambulab/BambuStudio #2897: Build instructions fails on both Mac and Linux

davglass, 2023-11-15:
### Bambu Studio Version

master

### Where is the application from?

Built from source by yourself or others

### OS version

OSX 14.2 Beta / Ubuntu 20

### Additional system information

Macbook Air M2
Lenovo Thinkpad i7 32GB

### Printer

X1C

### How to reproduce

Trying to build the deps from source, I get several errors on each OS, neither will finish the build so you can't build the app.

### Actual results

fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
20 errors generated.
20 errors generated.
20 errors generated.
20 errors generated.
make[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfFrameBuffer.cpp.o] Error 1
make[5]: *** Waiting for unfinished jobs....
make[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfChannelList.cpp.o] Error 1
make[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfAttribute.cpp.o] Error 1
make[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfBoxAttribute.cpp.o] Error 1
20 errors generated.
make[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfChannelListAttribute.cpp.o] Error 1
make[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfFloatAttribute.cpp.o] Error 1
20 errors generated.
make[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfCRgbaFile.cpp.o] Error 1
20 errors generated.
make[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfHeader.cpp.o] Error 1
make[4]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [dep_OpenEXR-prefix/src/dep_OpenEXR-stamp/dep_OpenEXR-build] Error 2
make[1]: *** [CMakeFiles/dep_OpenEXR.dir/all] Error 2
make: *** [all] Error 2

### Expected results

A working build system for the deps so you can compile the app.

### Project file & Debug log uploads

fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
20 errors generated.
20 errors generated.
20 errors generated.
20 errors generated.
make[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfFrameBuffer.cpp.o] Error 1
make[5]: *** Waiting for unfinished jobs....
make[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfChannelList.cpp.o] Error 1
make[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfAttribute.cpp.o] Error 1
make[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfBoxAttribute.cpp.o] Error 1
20 errors generated.
make[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfChannelListAttribute.cpp.o] Error 1
make[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfFloatAttribute.cpp.o] Error 1
20 errors generated.
make[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfCRgbaFile.cpp.o] Error 1
20 errors generated.
make[5]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/ImfHeader.cpp.o] Error 1
make[4]: *** [OpenEXR/IlmImf/CMakeFiles/IlmImf.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [dep_OpenEXR-prefix/src/dep_OpenEXR-stamp/dep_OpenEXR-build] Error 2
make[1]: *** [CMakeFiles/dep_OpenEXR.dir/all] Error 2
make: *** [all] Error 2

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-11-17:
@davglass 

we have tried building on macos and linux with no issue a few days agaon

could you give the full log of the building?

davglass, 2023-11-18:
Looks like there was an outage on one of the dependencies that get's downloaded as they now compiled without error. However, the main app does not:

```bash
cmake ..  -DBBL_RELEASE_TO_PUBLIC=1 -DCMAKE_PREFIX_PATH=/Users/davglass/src/bambu/deps/usr/local \
-DCMAKE_INSTALL_PREFIX="../install_dir" -DCMAKE_BUILD_TYPE=Release -DCMAKE_MACOSX_RPATH=ON \
-DCMAKE_INSTALL_RPATH=/Users/davglass/src/bambu/deps/usr/local -DCMAKE_MACOSX_BUNDLE=on
```

> -- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: OFF
-- SLIC3R_DESKTOP_INTEGRATION: ON
-- SLIC3R_ENC_CHECK: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: /Users/davglass/src/bambu/deps/usr/local (from cache or command line)
-- PROJECT_SOURCE_DIR: /Users/davglass/src/bambu/BambuStudio
-- CMAKE_MODULE_PATH: /Users/davglass/src/bambu/BambuStudio/cmake/modules/
OS X SDK Path: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk
OS X Deployment Target: 10.15
-- LIBDIR: /Users/davglass/src/bambu/BambuStudio/src
-- LIBDIR_BIN: /Users/davglass/src/bambu/BambuStudio/build/src
-- Found Boost: /Users/davglass/src/bambu/deps/usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found suitable version "1.78.0", minimum required is "1.66.0") found components: system filesystem thread log locale regex chrono atomic date_time iostreams 
-- Boost::boost exists
-- TBB found in /Users/davglass/src/bambu/deps/usr/local/lib/cmake/TBB
-- Found OpenSSL: /Users/davglass/src/bambu/deps/usr/local/lib/libcrypto.a (found version "3.1.2")  
-- Found OpenSSL: /Users/davglass/src/bambu/deps/usr/local/lib/libcrypto.a (found suitable version "3.1.2", minimum required is "3")  
-- Found ZLIB: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/lib/libz.tbd (found suitable version "1.2.12", minimum required is "1")  
-- Found ZLIB: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/usr/lib/libz.tbd (found version "1.2.12")  
-- FindGLEW: did not find GLEW CMake config file. Searching for libraries.
-- FindGLEW: Found OpenGL Framework.
-- FindGLEW: OPENGL_LIBRARIES: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/System/Library/Frameworks/OpenGL.framework;/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX14.0.sdk/System/Library/Frameworks/OpenGL.framework
-- FindGLEW: GLEW_USE_STATIC_LIBS: ON.
-- FindGLEW: GLEW_INCLUDE_DIR: /Users/davglass/src/bambu/deps/usr/local/include
-- FindGLEW: GLEW_INCLUDE_DIRS: /Users/davglass/src/bambu/deps/usr/local/include
-- FindGLEW: CMAKE_FIND_LIBRARY_SUFFIXES for SHARED: .dylib;.so
-- FindGLEW: CMAKE_FIND_LIBRARY_SUFFIXES for STATIC: .a
-- FindGLEW: GLEW_SHARED_LIBRARY_RELEASE: GLEW_SHARED_LIBRARY_RELEASE-NOTFOUND
-- FindGLEW: GLEW_STATIC_LIBRARY_RELEASE: /Users/davglass/src/bambu/deps/usr/local/lib/libGLEW.a
-- FindGLEW: GLEW_SHARED_LIBRARY_DEBUG: GLEW_SHARED_LIBRARY_DEBUG-NOTFOUND
-- FindGLEW: GLEW_STATIC_LIBRARY_DEBUG: GLEW_STATIC_LIBRARY_DEBUG-NOTFOUND
-- FindGLEW: GLEW_SHARED_LIBRARY: GLEW_SHARED_LIBRARY-NOTFOUND
-- FindGLEW: GLEW_STATIC_LIBRARY: /Users/davglass/src/bambu/deps/usr/local/lib/libGLEW.a
-- FindGLEW: GLEW_LIBRARIES: /Users/davglass/src/bambu/deps/usr/local/lib/libGLEW.a
-- FindGLEW: GLEW_VERSION_MAJOR: 2
-- FindGLEW: GLEW_VERSION_MINOR: 2
-- FindGLEW: GLEW_VERSION_MICRO: 0
-- FindGLEW: GLEW_VERSION: 2.2.0
-- FindGLEW: Creating GLEW::glew_s imported target.
-- FindGLEW: Creating GLEW::GLEW imported target.
-- Found NLopt in '/Users/davglass/src/bambu/deps/usr/local/lib'.
-- Using NLopt include directory '/Users/davglass/src/bambu/deps/usr/local/include'.
-- Using NLopt library '/Users/davglass/src/bambu/deps/usr/local/lib/libnlopt.a'.
CMake Deprecation Warning at cmake/modules/FindOpenVDB.cmake:126 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.
  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
Call Stack (most recent call first):
  CMakeLists.txt:549 (find_package)
-- OpenVDB ABI Version: 8
[... 100 lines cut ...]
[  1%] Built target global-encoding-check
[  2%] Building CXX object src/admesh/CMakeFiles/admesh.dir/connect.cpp.o
In file included from /Users/davglass/src/bambu/BambuStudio/src/admesh/connect.cpp:32:
In file included from /Users/davglass/src/bambu/deps/usr/local/include/boost-1_78/boost/log/trivial.hpp:23:
In file included from /Users/davglass/src/bambu/deps/usr/local/include/boost-1_78/boost/log/sources/severity_logger.hpp:23:
In file included from /Users/davglass/src/bambu/deps/usr/local/include/boost-1_78/boost/log/sources/basic_logger.hpp:36:
In file included from /Users/davglass/src/bambu/deps/usr/local/include/boost-1_78/boost/log/core/core.hpp:23:
In file included from /Users/davglass/src/bambu/deps/usr/local/include/boost-1_78/boost/log/core/record.hpp:21:
In file included from /Users/davglass/src/bambu/deps/usr/local/include/boost-1_78/boost/log/attributes/attribute_value_set.hpp:27:
[... cut ...]
using __unary_function = __unary_function_keep_layout_base<_Arg, _Result>;
^
/Users/davglass/src/bambu/BambuStudio/src/admesh/connect.cpp:219:16: warning: unused variable 'temp' [-Wunused-variable]
                                        HashEdge *temp = link->next;
                                                  ^
1 warning and 1 error generated.
make[2]: *** [src/admesh/CMakeFiles/admesh.dir/connect.cpp.o] Error 1
make[1]: *** [src/admesh/CMakeFiles/admesh.dir/all] Error 2
make: *** [all] Error 2

lanewei120, 2023-11-18:
this is caused by the xcode sdk's update
your xcode should be a new version

we can find lots of similiar issues:
https://github.com/facebook/react-native/issues/37748
https://stackoverflow.com/questions/77133361/no-template-named-unary-function-in-namespace-std-did-you-mean-unary-fun
https://forum.mendix.com/link/space/mobile/questions/128794

currently you can either do following things to fix
1. replace unary_function with __unary_function  in /Users/davglass/src/bambu/deps/usr/local/include/boost-1_78/boost/container_hash/hash.hpp and go on building
or
2. add compiling flag _LIBCPP_ENABLE_CXX17_REMOVED_UNARY_BINARY_FUNCTION into CMakeLists.txt , for example
diff --git a/CMakeLists.txt b/CMakeLists.txt
index f9b17ee55..b806bb0f3 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -275,8 +275,8 @@ if (SLIC3R_ASAN)
 endif ()

 if (APPLE)
-    set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Werror=partial-availability -Werror=unguarded-availability -Werror=unguarded-availability-new")
-    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Werror=partial-availability -Werror=unguarded-availability -Werror=unguarded-availability-new")
+    set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Werror=partial-availability -Werror=unguarded-availability -Werror=unguarded-availability-new /D_LIBCPP_ENABLE_CXX17_REMOVED_UNARY_BINARY_FUNCTION")
+    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Werror=partial-availability -Werror=unguarded-availability -Werror=unguarded-availability-new /D_LIBCPP_ENABLE_CXX17_REMOVED_UNARY_BINARY_FUNCTION")
 endif ()

 if(MSVC)

we will fix it in future version

thanks

closed 2025-04-29 by MackBambu