bambulab/BambuStudio #2899: Windows 11 Building Failed！

zhangjiangzhao123, 2023-11-16:
### Bambu Studio Version

1.7.2

### Where is the application from?

Built from source by yourself or others

### OS version

windows11

### Additional system information

13th Gen Intel(R) Core(TM) i5-13500H   2.60 GHz
16.0 GB (15.7 GB 可用)

### Printer

无

### How to reproduce

使用x86 Native Tools Command Prompt for VS 2022作为windows shell
1. cmake ../ -G "Visual Studio 17 2022" -DDESTDIR="D:/Project/BambuStudio/BambuStudio_deps" -DCMAKE_BUILD_TYPE=Release 
2. msbuild /m ALL_BUILD.vcxproj

create a directory to store the dependence built "D:/Project/BambuStudio/BambuStudio_deps" 
当生成dep_wxWidgets时出错，无论我使用VS打开工程选择release编译还是使用上述代码编译，均出现dep_wxWidgets相关的错误，似乎是无法匹配。

### Actual results

       -- Configured wxWidgets 3.1.5 for Windows-10.0.22621
             Min OS Version required at runtime:                Windows Vista / Windows Server 2008 (x64 Edition)
             Which GUI toolkit should wxWidgets use?            msw
             Should wxWidgets be compiled into single library?  OFF
             Should wxWidgets be linked as a shared library?    OFF
             Should wxWidgets support Unicode?                  ON
             What wxWidgets compatibility level should be used? 3.0
         -- Configuring incomplete, errors occurred!
    17>C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(24
       8,5): error MSB8066: “D:\Project\BambuStudio\deps\build\CMakeFiles\11e30cd00a09f964c70b7fb45641565b\dep_wxWidget
       s-mkdir.rule;D:\Project\BambuStudio\deps\build\CMakeFiles\11e30cd00a09f964c70b7fb45641565b\dep_wxWidgets-downloa
       d.rule;D:\Project\BambuStudio\deps\build\CMakeFiles\11e30cd00a09f964c70b7fb45641565b\dep_wxWidgets-update.rule;D
       :\Project\BambuStudio\deps\build\CMakeFiles\11e30cd00a09f964c70b7fb45641565b\dep_wxWidgets-patch.rule;D:\Project
       \BambuStudio\deps\build\CMakeFiles\11e30cd00a09f964c70b7fb45641565b\dep_wxWidgets-configure.rule;D:\Project\Bamb
       uStudio\deps\build\CMakeFiles\11e30cd00a09f964c70b7fb45641565b\dep_wxWidgets-build.rule;D:\Project\BambuStudio\d
       eps\build\CMakeFiles\11e30cd00a09f964c70b7fb45641565b\dep_wxWidgets-build_debug.rule;D:\Project\BambuStudio\deps
       \build\CMakeFiles\11e30cd00a09f964c70b7fb45641565b\dep_wxWidgets-install.rule;D:\Project\BambuStudio\deps\build\
       CMakeFiles\8eb0102a5685fa539f60bbeb70badc3b\dep_wxWidgets-complete.rule;D:\Project\BambuStudio\deps\build\CMakeF
       iles\f39e8daba1f2bddd7251acf95db380db\dep_wxWidgets.rule;D:\Project\BambuStudio\deps\CMakeLists.txt”的自定义生成 已退出，代码
       为 1。 [D:\Project\BambuStudio\deps\build\dep_wxWidgets.vcxproj]
    17>已完成生成项目“D:\Project\BambuStudio\deps\build\dep_wxWidgets.vcxproj”(默认目标)的操作 - 失败。
     3>已完成生成项目“D:\Project\BambuStudio\deps\build\deps.vcxproj”(默认目标)的操作 - 失败。
     1>已完成生成项目“D:\Project\BambuStudio\deps\build\ALL_BUILD.vcxproj”(默认目标)的操作 - 失败。

### Expected results

成功生成依赖项

### Project file & Debug log uploads

         CUSTOMBUILD : warning : 7 lines applied after fixing whitespace errors. [D:\Project\BambuStudio\deps\build\dep
       _wxWidgets.vcxproj]

       “D:\Project\BambuStudio\deps\build\ALL_BUILD.vcxproj”(默认目标) (1) ->
       “D:\Project\BambuStudio\deps\build\deps.vcxproj”(默认目标) (3) ->
       “D:\Project\BambuStudio\deps\build\dep_OpenVDB.vcxproj”(默认目标) (18) ->
       “D:\Project\BambuStudio\deps\build\dep_OpenEXR.vcxproj”(默认目标) (24) ->
       (CustomBuild 目标) ->
         CUSTOMBUILD : error : corrupt patch at line 5 [D:\Project\BambuStudio\deps\build\dep_OpenEXR.vcxproj]
         C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(
       248,5): error MSB8066: “D:\Project\BambuStudio\deps\build\CMakeFiles\b978c09b84f7ff6a0ef8d338473a2780\dep_OpenEX
       R-mkdir.rule;D:\Project\BambuStudio\deps\build\CMakeFiles\b978c09b84f7ff6a0ef8d338473a2780\dep_OpenEXR-download.
       rule;D:\Project\BambuStudio\deps\build\CMakeFiles\b978c09b84f7ff6a0ef8d338473a2780\dep_OpenEXR-update.rule;D:\Pr
       oject\BambuStudio\deps\build\CMakeFiles\b978c09b84f7ff6a0ef8d338473a2780\dep_OpenEXR-patch.rule;D:\Project\Bambu
       Studio\deps\build\CMakeFiles\b978c09b84f7ff6a0ef8d338473a2780\dep_OpenEXR-configure.rule;D:\Project\BambuStudio\
       deps\build\CMakeFiles\b978c09b84f7ff6a0ef8d338473a2780\dep_OpenEXR-build.rule;D:\Project\BambuStudio\deps\build\
       CMakeFiles\b978c09b84f7ff6a0ef8d338473a2780\dep_OpenEXR-build_debug.rule;D:\Project\BambuStudio\deps\build\CMake
       Files\b978c09b84f7ff6a0ef8d338473a2780\dep_OpenEXR-install.rule;D:\Project\BambuStudio\deps\build\CMakeFiles\8eb
       0102a5685fa539f60bbeb70badc3b\dep_OpenEXR-complete.rule;D:\Project\BambuStudio\deps\build\CMakeFiles\f39e8daba1f
       2bddd7251acf95db380db\dep_OpenEXR.rule;D:\Project\BambuStudio\deps\CMakeLists.txt”的自定义生成已退出，代码为 128 。 [D:\Project
       \BambuStudio\deps\build\dep_OpenEXR.vcxproj]

       “D:\Project\BambuStudio\deps\build\ALL_BUILD.vcxproj”(默认目标) (1) ->
       “D:\Project\BambuStudio\deps\build\deps.vcxproj”(默认目标) (3) ->
       “D:\Project\BambuStudio\deps\build\dep_wxWidgets.vcxproj”(默认目标) (17) ->
         C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(
       248,5): error MSB8066: “D:\Project\BambuStudio\deps\build\CMakeFiles\11e30cd00a09f964c70b7fb45641565b\dep_wxWidg
       ets-mkdir.rule;D:\Project\BambuStudio\deps\build\CMakeFiles\11e30cd00a09f964c70b7fb45641565b\dep_wxWidgets-downl
       oad.rule;D:\Project\BambuStudio\deps\build\CMakeFiles\11e30cd00a09f964c70b7fb45641565b\dep_wxWidgets-update.rule
       ;D:\Project\BambuStudio\deps\build\CMakeFiles\11e30cd00a09f964c70b7fb45641565b\dep_wxWidgets-patch.rule;D:\Proje
       ct\BambuStudio\deps\build\CMakeFiles\11e30cd00a09f964c70b7fb45641565b\dep_wxWidgets-configure.rule;D:\Project\Ba
       mbuStudio\deps\build\CMakeFiles\11e30cd00a09f964c70b7fb45641565b\dep_wxWidgets-build.rule;D:\Project\BambuStudio
       \deps\build\CMakeFiles\11e30cd00a09f964c70b7fb45641565b\dep_wxWidgets-build_debug.rule;D:\Project\BambuStudio\de
       ps\build\CMakeFiles\11e30cd00a09f964c70b7fb45641565b\dep_wxWidgets-install.rule;D:\Project\BambuStudio\deps\buil
       d\CMakeFiles\8eb0102a5685fa539f60bbeb70badc3b\dep_wxWidgets-complete.rule;D:\Project\BambuStudio\deps\build\CMak
       eFiles\f39e8daba1f2bddd7251acf95db380db\dep_wxWidgets.rule;D:\Project\BambuStudio\deps\CMakeLists.txt”的自定义生 成已退出，
       代码为 1。 [D:\Project\BambuStudio\deps\build\dep_wxWidgets.vcxproj]

    2030 个警告
    3 个错误

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhangjiangzhao123, 2023-11-17:
[错误log.txt](https://github.com/bambulab/BambuStudio/files/13386548/log.txt)
附件上传了完整的日志

bambu123, 2023-11-17:
请清空 build 目录， 重试，然后给我们全新构建的错误日志。
因为要执行的 patch 不能二次应用，不能支持同一个 build 目录二次构建。

zhangjiangzhao123, 2023-11-17:
[cmd.txt](https://github.com/bambulab/BambuStudio/files/13388634/cmd.txt)
[myLog.txt](https://github.com/bambulab/BambuStudio/files/13388636/myLog.txt)
我重新清空build目录以及BambuStudio_deps目录下的所有文件和文件夹，重新生成，还是遇到问题，cmd上的输入已经在附件cmd.txt中，而cmd的输出日志全部记录在附件myLog.txt中。
失败也许是我一个很小的疏忽导致的，但很遗憾我没看出来。希望能解决我的问题，非常感谢。

bambu123, 2024-02-06:
CMake Error at build/cmake/lib/webview/CMakeLists.txt:78 (file):
           file DOWNLOAD HASH mismatch
         
             for file: [D:/Project/BambuStudio/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets-build/webview2.nuget]
               expected hash: [6a34bb553e18cfac7297b4031f3eac2558e439f8d16a45945c22945ac404105d]
                 actual hash: [51d2ef56196e2a9d768a6843385bcb9c6baf9ed34b2603ddb074fb4995543a99]

         CUSTOMBUILD : error : corrupt patch at line 5 [D:\Project\BambuStudio\deps\build\dep_OpenEXR.vcxproj]

zhangjiangzhao123, 2024-02-18:
所以，如何解决这一问题？

bambu123, 2024-02-20:
下载出问题了，需要好的网络，再重试。

stxve4203, 2024-02-23:
I have the same issue, I cant compile the bambuslicer neither my own with webview2, I´ve added all the including stuff but it´s always crashing or not finish compiling.

frank-pian, 2024-03-02:
类似的错误，全局挂了代理，没有网络问题，删除重新构建了几次，每次都会报错
```
“E:\BambuStudio\deps\build\ALL_BUILD.vcxproj”(默认目标) (1) ->
       “E:\BambuStudio\deps\build\deps.vcxproj”(默认目标) (3) ->
       “E:\BambuStudio\deps\build\dep_CURL.vcxproj”(默认目标) (7) ->
       (CustomBuild 目标) ->
         C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(
       254,5): error MSB8066: “E:\BambuStudio\deps\build\CMakeFiles\1cb165292839f72491f0dff422977af8\dep_CURL-mkdir.rul
       e;E:\BambuStudio\deps\build\CMakeFiles\1cb165292839f72491f0dff422977af8\dep_CURL-download.rule;E:\BambuStudio\de
       ps\build\CMakeFiles\1cb165292839f72491f0dff422977af8\dep_CURL-update.rule;E:\BambuStudio\deps\build\CMakeFiles\1
       cb165292839f72491f0dff422977af8\dep_CURL-patch.rule;E:\BambuStudio\deps\build\CMakeFiles\1cb165292839f72491f0dff
       422977af8\dep_CURL-configure.rule;E:\BambuStudio\deps\build\CMakeFiles\1cb165292839f72491f0dff422977af8\dep_CURL
       -build.rule;E:\BambuStudio\deps\build\CMakeFiles\1cb165292839f72491f0dff422977af8\dep_CURL-build_debug.rule;E:\B
       ambuStudio\deps\build\CMakeFiles\1cb165292839f72491f0dff422977af8\dep_CURL-install.rule;E:\BambuStudio\deps\buil
       d\CMakeFiles\51a6e2c503bb5c7e6143482b2d0e673c\dep_CURL-complete.rule;E:\BambuStudio\deps\build\CMakeFiles\6def94
       6b31b3e18f4ac8ba41f27929a5\dep_CURL.rule;E:\BambuStudio\deps\CMakeLists.txt”的自定义生成已退出，代码为 1。 [E:\BambuStudio\dep
       s\build\dep_CURL.vcxproj]

       “E:\BambuStudio\deps\build\ALL_BUILD.vcxproj”(默认目标) (1) ->
       “E:\BambuStudio\deps\build\deps.vcxproj”(默认目标) (3) ->
       “E:\BambuStudio\deps\build\dep_wxWidgets.vcxproj”(默认目标) (24) ->
       “E:\BambuStudio\deps\build\dep_TIFF.vcxproj”(默认目标) (26) ->
         E:\BambuStudio\deps\build\dep_TIFF-prefix\src\dep_TIFF\tools\tiffgt.c(45,11): error C1083: 无法打开包括文件: “GL/glut.
       h”: No such file or directory [E:\BambuStudio\deps\build\dep_TIFF-prefix\src\dep_TIFF-build\tools\tiffgt.vcxproj
       ] [E:\BambuStudio\deps\build\dep_TIFF.vcxproj]
         C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(
       254,5): error MSB8066: “E:\BambuStudio\deps\build\CMakeFiles\88d6e35fbda6f6f5265694a2ef5aed5f\dep_TIFF-mkdir.rul
       e;E:\BambuStudio\deps\build\CMakeFiles\88d6e35fbda6f6f5265694a2ef5aed5f\dep_TIFF-download.rule;E:\BambuStudio\de
       ps\build\CMakeFiles\88d6e35fbda6f6f5265694a2ef5aed5f\dep_TIFF-update.rule;E:\BambuStudio\deps\build\CMakeFiles\8
       8d6e35fbda6f6f5265694a2ef5aed5f\dep_TIFF-patch.rule;E:\BambuStudio\deps\build\CMakeFiles\88d6e35fbda6f6f5265694a
       2ef5aed5f\dep_TIFF-configure.rule;E:\BambuStudio\deps\build\CMakeFiles\88d6e35fbda6f6f5265694a2ef5aed5f\dep_TIFF
       -build.rule;E:\BambuStudio\deps\build\CMakeFiles\88d6e35fbda6f6f5265694a2ef5aed5f\dep_TIFF-install.rule;E:\Bambu
       Studio\deps\build\CMakeFiles\51a6e2c503bb5c7e6143482b2d0e673c\dep_TIFF-complete.rule;E:\BambuStudio\deps\build\C
       MakeFiles\6def946b31b3e18f4ac8ba41f27929a5\dep_TIFF.rule;E:\BambuStudio\deps\CMakeLists.txt”的自定义生成已退出，代码为 1。 [E:
       \BambuStudio\deps\build\dep_TIFF.vcxproj]
```

buchankn, 2024-12-07:
I also have build errors with Visual Studio 2022. I investigated this some more, and it looks like the version of Boost libraries that BambuStudio builds with won't build with Visual Studio 2022. The msvc.jam file in Boost version 1.78 doesn't have any configuration for Visual Studio 2022.

It looks like Boost has to be upgraded to a newer version in the build.

MackBambu, 2025-03-12:
旧的编译指南确实有些问题，刚刚调整了编译命令请测试一下
https://github.com/bambulab/BambuStudio/wiki/Windows-Compile-Guide

构建指令可以参考.github下的build_deps.yml和build_bambu.yml，它们都是保持正确的并且在github ci进行了验证。

MackBambu, 2025-03-12:
The old compilation guide did have some issues. I have just updated the build commands, please test them:
[Windows Compile Guide](https://github.com/bambulab/BambuStudio/wiki/Windows-Compile-Guide).

You can refer to the build instructions in .github/build_deps.yml and .github/build_bambu.yml. These are kept correct and have been verified in GitHub CI.

vanowm, 2025-07-28:
> The old compilation guide did have some issues. I have just updated the build commands, please test them: [Windows Compile Guide](https://github.com/bambulab/BambuStudio/wiki/Windows-Compile-Guide).

There are some discrepancies in the guide:

> open 'x64 Native Tools Command Prompt for VS 2019' as the windows shell

Shouldn't it be?
> open 'x64 Native Tools Command Prompt for VS 2022' as the windows shell

Another:
>create a directory to store the dependence built: D:/work/Projects/BambuStudio_dep

but later it says:
>cd BambuStudio/deps
mkdir build
mkdir BambuStudio_dep
cd build
cmake ../ -G "Visual Studio 17 2022" -A x64 -DDESTDIR="./BambuStudio_dep" -DCMAKE_BUILD_TYPE=Release -DDEP_DEBUG=OFF
msbuild /m ALL_BUILD.vcxproj

And not a single command is using `D:/work/Projects/BambuStudio_dep`

I was not able build from source, perhaps the required software is newer and no longer compatible with the build process. If that's the case, please provide which version of additional software is required.