bambulab/BambuStudio #2900: Switching Machine Preset Changes Filament and Process Presets

CCS86, 2023-11-16:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab github releases

### OS version

Win 10 x64

### Printer

X1C

### How to reproduce

1. Make and save a new user machine preset
2. Start a new project
3. Create a new user process preset
4. Get multiple color user presets selected
5. Switch the machine preset

### Actual results

When you select a different machine preset, the process preset and the filament presets will change, erasing any part level multi-color selections you had made:

[image]

### Expected results

I expect to be able to change the machine preset without affecting the process or filament presets.

### Project file & Debug log uploads

...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2023-11-17:
it's Orca. Please check if bambuStudio has the same issue or not. if it has, you can update the information and upload 3mf, we will check it.

CCS86, 2023-11-17:
> it's Orca. Please check if bambuStudio has the same issue or not. if it has, you can update the information and upload 3mf, we will check it.

Sorry about that, I thought the behavior was more global, but I haven't reproduced in BS yet. I'll close the issue until I see it in BS.

closed 2023-11-17 by CCS86