bambulab/BambuStudio #2901: Project Tab

pvranch, 2023-11-16:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS: 13.1 (22C65)

### Printer

Bambu X1 Carbon

### How to reproduce

The projects tab no longer contains information I can add or edit. I want to know why it is missing and if it will be fixed. This is where I keep vital information for projects and job-related information.

### Actual results

The projects tab no longer contains information I can add or edit. I want to know why it is missing and if it will be fixed. This is where I keep vital information for projects and job-related information.

### Expected results

The projects tab no longer contains information I can add or edit. I want to know why it is missing and if it will be fixed. This is where I keep vital information for projects and job-related information.

### Project file & Debug log uploads

The projects tab no longer contains information I can add or edit. I want to know why it is missing and if it will be fixed. This is where I keep vital information for projects and job-related information.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2023-11-17:
it is moved to the page where you share your project on the makerworld.

pvranch, 2023-11-18:
I don't want to share some of my projects on the makerworld. They are for me only.

mrsgit09, 2023-11-18:
+1 for pvranch's comment. There are LOTs of projects I don't want to share on Makerworld. This is a regression in my opinion.

pvranch, 2023-11-18:
You could put it back the way it was, and if we want to share it on Makerworld, put a check box to share it. That would work for me. Many of us like to share projects, which is great, but some, if not many, are designed to make money in our business.

walterwongbbl, 2023-12-21:
If you have any other issues, please create a new issue. The current conversation will be closed.

closed 2023-12-21 by walterwongbbl