bambulab/BambuStudio #2902: BambuSlicer CLI does not seem to correctly honor `--arrange 1`

eropple, 2023-11-16:
### Bambu Studio Version

v01.08.00.57

### Where is the application from?

Bambu Lab github releases

### OS version

Fedora 39

### Additional system information

CPU: Ryzen 3900x
RAM: 64GB DDR4-3000
GPU: nVidia 2070 Super

### Printer

n/a

### How to reproduce

My first comment to this bug report will contain a reproduction sample, including scripts to run it.

### Actual results

Models seem to be adequately (but not efficiently) packed onto the first plate in the 3MF, but models that don't fit in the naive case are put off the plate and no second plates are then created.

### Expected results

I expect models to arrange similarly to the GUI slicer, creating new plates as necessary.

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [X] Log file
- [X] Project file

eropple, 2023-11-16:
[bugrep.zip](https://github.com/bambulab/BambuStudio/files/13379773/bugrep.zip) contains an example and the necessary scripts to run it. I did consider that repeated filenames might be an issue, and tested against it; it is **not** the issue at hand here; with separate model files, BambuStudio's CLI created the same output as the images below. (I omitted the second test because GitHub has a 25MB file size limit.)

This is a pretty hard blocker for me and it's rendering my printers unusable in my current workflow; advice would be appreciated.

Generated:
[image]

Expected:
[image]

eropple, 2023-11-17:
I've also added a log here: https://gist.github.com/eropple/e4c7a4ce4bd26066c18c8ea951e3ae70

In particular, something interesting -- the arrange code appears to be adding extra plates as one would expect, but doesn't appear to be arranging onto them correctly, so that the plates are then deleted because they have no models on them.

eropple, 2023-11-22:
Hi - we're approaching a week on this without an ack, aside from being assigned to someone. When can I expect some insight around prioritization and time to delivery?

lanewei120, 2023-11-22:
what is your CLI command?

the arrange under CLI has not been fully tested yet
could you also show us your detailed requirement?

eropple, 2023-11-22:
Hi @lanewei120 - the CLI command is in `bugrep.zip`, provided in [this comment](https://github.com/bambulab/BambuStudio/issues/2902#issuecomment-1814737930). You can run this and reproduce exactly my test case.

I'll include it here for convenience, but I'd recommend looking at the zip file and its contents.

Thanks!

```bash
#! /bin/bash

# bail if BAMBUSTUDIO_PATH is not set
if [ -z "$BAMBUSTUDIO_PATH" ]; then
    echo "BAMBUSTUDIO_PATH is not set"
    exit 1
fi

"$BAMBUSTUDIO_PATH" \
    --debug 5 \
    --load-settings "machine_profile.json;process_profile.json" \
    --export-3mf "./GENERATED.3mf" \
    --arrange=1 \
    3DBenchy.stl \
    3DBenchy.stl \
    3DBenchy.stl \
    3DBenchy.stl \
    BigCube.stl \
    BigCube.stl \
    BigCube.stl \
    notsphere.stl \
    Cube.stl \
    Cube.stl \
    Cube.stl \
    Voron_Design_Cube_v7.stl
```

lanewei120, 2023-11-22:
are you using the newest codes on the master to do the testing?
I just tried your command, and it seems the result is correct here
[image]

./bin/bambu-studio --debug 3 --export-3mf out.3mf --load-settings "machine.json;process.json" --arrange=1 ~/temp/test1-repeated_files/3DBenchy.stl ~/temp/test1-repeated_files/3DBenchy.stl ~/temp/test1-repeated_files/3DBenchy.stl ~/temp/test1-repeated_files/3DBenchy.stl ~/temp/test1-repeated_files/BigCube.stl ~/temp/test1-repeated_files/BigCube.stl ~/temp/test1-repeated_files/BigCube.stl ~/temp/test1-repeated_files/notsphere.stl ~/temp/test1-repeated_files/Cube.stl ~/temp/test1-repeated_files/Cube.stl ~/temp/test1-repeated_files/Cube.stl ~/temp/test1-repeated_files/Voron_Design_Cube_v7.stl

lanewei120, 2023-11-22:
I also add one more big cube
[image]
the second plate is generated successfully

please try it with the newest codes on the master branch

eropple, 2023-11-23:
@lanewei120 Unfortunately, I'm still not able to replicate what you're seeing.

```bash
$ for branch in `git branch -r | grep -v HEAD`;do echo -e `git show --format="%ci %cr" $branch | head -n 1` \\t$branch; done | sort -r
2023-11-22 17:46:20 +0800 15 hours ago  personal/master
2023-11-22 17:46:20 +0800 15 hours ago  origin/master
2023-11-09 14:26:03 +0800 2 weeks ago   origin/github_master
2023-10-30 08:43:52 +0800 3 weeks ago   origin/v1.7.2_1016
2023-10-13 14:43:42 +0800 6 weeks ago   origin/v1.7.2
2023-04-28 17:11:46 +0800 7 months ago  origin/v1.6.0
2023-03-14 20:52:40 +0800 8 months ago  origin/v1.5.0
2023-02-08 11:19:32 +0800 10 months ago         origin/v1.4.2
2022-12-20 22:12:34 +0800 11 months ago         origin/v1.4.0

$ # compiled the application here

$ ~/path/to/BambuStudio/build/src/bambu-studio --debug 3 --export-3mf GENERATED6.3mf --load-settings "machine_profile.json;process_profile.json" --arrange=1 ./3DBenchy.stl ./3DBenchy.stl ./3DBenchy.stl ./3DBenchy.stl ./BigCube.stl ./BigCube.stl ./BigCube.stl ./BigCube.stl ./BigCube.stl ./notsphere.stl ./Cube.stl ./Cube.stl ./Cube.stl ./Voron_Design_Cube_v7.stl ./BigCube.stl ./Voron_Design_Cube_v7.stl ./Voron_Design_Cube_v7.stl ./Voron_Design_Cube_v7.stl

# output is in gist: https://gist.github.com/eropple/4272450c5fb9fb8451eeb26f351ddd8d

```

You can see the output from the command [here](https://gist.github.com/eropple/4272450c5fb9fb8451eeb26f351ddd8d). But here's a screenshot when I open it in the same `master` build of Bambu Studio:

[image]

That being said, when I opened the 3mf file in Bambu Studio, there is a flash of what looks like three bed layouts (horizontally laid) before it flashes to this arrangement. Three beds would also not be 'correct', but would be more correct than what we're seeing here. I'm _not_ sure this is meaningful, however, because when I attempt to pass `--slice=0` on the command line, I get the following `result.json`:

```json
$ cat result.json
{
    "error_string": "Some objects are located over the boundary of the heated bed.",
    "export_time": 0,
    "plate_index": 1,
    "prepare_time": 1,
    "return_code": -52
}
```

And that matches what I'm seeing visually in the screenshot - the various cubes are overlaid over the edge of the bed. I unzipped the 3MF file to check its contents, and that also only shows two plates and models in strange places as well.

I considered that there might be a material difference in the dependencies available in Fedora 39 and Ubuntu 23.10, too. But that would be really alarming. Just to be _certain_, I created an Ubuntu 22.04 VM and `master` exhibited the same behavior, as per:

[image]

I dunno, dude - the only thing I can think of is that maybe _you're_ running code that isn't in public `master`? If you could upload the AppImage you've built from your local, I could try that.

As an aside: it might be a good idea to add a GitHub Action for pushes/merges to `master` that can build every platform so that we can have unambiguous versions to refer to and test against.

eropple, 2023-11-23:
Actually, one thing @lanewei120 -- I did notice that you referred to `process.json` and `machine.json`, not my `machine_profile.json` and `process_profile.json`. Do those files differ from the ones in bugrep.zip? If so, can you post them?

lanewei120, 2023-11-23:
[machine.json](https://github.com/bambulab/BambuStudio/files/13446026/machine.json)
[process.json](https://github.com/bambulab/BambuStudio/files/13446027/process.json)
sure
please have a try

eropple, 2023-11-23:
@lanewei120 Interesting! I can repro with your process file, and my machine file (a P1S rather than an X1C). Mine (copied directly out of Bambu Studio) refers to `"inherits": "fdm_process_bbl_0.20"`; yours does as well, but looks like it fully incorporates `fdm_process_bbl_0.20.json` and its inheritance tree in it.

Is it just not following the inheritance tree and hitting an edge case in the arrange logic? Does a profile need to be fully established, with all properties pulled in, for Bambu Studio to understand it?

eropple, 2023-11-23:
(I think this might be related to #2940, by the way)

lanewei120, 2023-11-23:
there are some options in process.json which will affect the arrange,
we need to pass a full profile to CLI, the CLI won't read the inherits like the GUI program does

eropple, 2023-11-23:
That makes sense - would strongly suggest some sort of error check to kick
back if it isn't a full profile, because you can't really *get* a full
profile without doing some work.

lanewei120, 2023-11-24:
we will add it into the  wiki page of CLI

lanewei120, 2024-02-18:
already added into wiki before, closed

closed 2024-02-18 by lanewei120