bambulab/BambuStudio #2903: [v1.8 BETA] Allow conversion of old custom Filaments

Pmant, 2023-11-16:
**Is your feature request related to a problem? Please describe.**
I had a bunch of custom Filaments in Bambu Slicer prior to 1.8.0. I just started with a Generic profile (e.g. Generic TPU) and edited the profile to my needs, changed the Vendor Name. Example:
[image]

These profiles are still available in 1.8.0 but are not recognized as Custom Filaments, furthermore the Vendor is not an editable field anymore. When using the wizard to create new profiles I also cannot use these profiles as source.

**Describe the solution you'd like**
Either automatically convert 1.7.x profiles with custom vendor names to custom profiles in 1.8.0 or allow to convert them manually.

QingZhangBambu, 2023-11-17:
i will talk to concerned staff about that.

DanBao-Bambu, 2023-11-17:
Hello.
The Create Filament feature allows for the conversion of previous custom presets to third-party consumable presets.
The steps are as follows:
[image]
After filling in the basic information, choose to create based on Filament Preset.
Select a custom configuration file that you want to convert. Click Create to succeed.
Note: Custom presets of the same type can only create one Filament under the same printer. If you have multiple custom presets of the same type, you can enter different serials to distinguish them.

Pmant, 2023-11-17:
Thank you for the quick response.  This list does not show my old custom filaments:
[image]

I noticed that beta 1.8.0 has its own settings distinct from 1.7.x and the profiles imported via cloud sync into 1.8.0. Maybe this helps debugging the issue.

DanBao-Bambu, 2023-11-18:
Sorry, this is a known issue. The current 1.8Beta version relies on names to filter presets, and your preset name does not have a TPU, so it cannot be displayed. The next version will solve your problem. If you are in a hurry, you can first change the name of your preset to include the TPU field.

Pmant, 2023-11-18:
That solves the issue, thank you.

DanBao-Bambu, 2023-11-20:
Sorry, this issue is known and will be resolved in the next version.

Pmant, 2023-11-23:
This is solved now if the old filament does have the filament type in it's name.

closed 2023-11-23 by Pmant