bambulab/BambuStudio #2904: Not showing text of design & wrong amount of layers

NVNDO, 2023-11-16:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: Intel  i7 7700K
RAM: 16GB DDR4 CL14
GPU: Nvidia RTX 2060

### Printer

Bambulabs P1S

### How to reproduce

Problem Nr.1:  Create a coaster with text (as separate body) 0.4mm deep in Fusion 360 ->Import design into BambuStudio

[image]
[image]

Problem Nr.2: In "Preparemode" change color nr. 1 to white to make the entire coaster white. 
- Now paint only the coaster (top/side/bottom) and remaining spots in black with the Fill-tool.
- Go to "Preview"

[image]

### Actual results

Problem Nr.1: TEXT NOT VISIBLE - Neither in "Prepare" nor "Preview"

[image]
[image]
[image]

Problem Nr.2: In "Preview" 4 layers of text will be shown in white
[image]
[image]
[image]
[image]
[image]
[image]

### Expected results

Problem Nr.1: Preview of design should show entirely - with its text (as its part of the design)

Problem Nr.2: Only 2 layers of text should be shown in white as the designed text is only 0.4mm hight/deep and the layer height is set to 0.2mm (0.2+0.2=0.4 = 2 layer)

### Project file & Debug log uploads

[BBL NOT showing Text.zip](https://github.com/bambulab/BambuStudio/files/13384930/BBL.NOT.showing.Text.zip)

[debug_Fri_Nov_17_00_05_12_22212.log.zip](https://github.com/bambulab/BambuStudio/files/13384931/debug_Fri_Nov_17_00_05_12_22212.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2023-11-24:
[image]
@NVNDO  I used version V1.8.0.57 to open the 3mf file you provided. The text part can be displayed, and the slices can also be displayed normally. I have not reproduced your problem.

NVNDO, 2023-11-28:
Hey @zhimin-zeng-bambulab, thx for checking that, I found out about that release of 1.8.. after publishing my request.
As it was not the Official release on your website I wasn't sure if that was already the one to go. 

I'll downloaded it later this day and see how it goes - but by seeing your post, the problem seems to be resolved. 
Thank you!