bambulab/BambuStudio #2905: BBLStudio creates more layers of text than actually exist

NVNDO, 2023-11-16:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: Intel i7 7700K
RAM: 16GB DDR4 CL14
GPU: Nvidia RTX 2060

### Printer

Bambulabs P1S

### How to reproduce

1. Create a Coaster with 0.4mm deep text in Fusion 360 -> Import into BambuLabsStudio

[image]
[image]

2. In "Prepare-mode" change color nr. 1 to white to make the entire coaster white. 
- Now paint only the coaster (top/side/bottom) and remaining spots in black with the Fill-tool.
- Go to "Preview"

[image]

### Actual results

In "Preview" 4 layers of text will be shown in white
[image]
[image]
[image]
[image]
[image]
[image]

### Expected results

Only 2 layers of text should be shown in white as the designed text is only 0.4mm high/deep and the layer height is set to 0.2mm (0.2+0.2=0.4 = 2 layer)

### Project file & Debug log uploads

[BBL NOT showing Text.zip](https://github.com/bambulab/BambuStudio/files/13385062/BBL.NOT.showing.Text.zip)
[debug_Fri_Nov_17_00_05_12_22212.log.zip](https://github.com/bambulab/BambuStudio/files/13385063/debug_Fri_Nov_17_00_05_12_22212.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2023-11-24:
@NVNDO 
After converting to a 3mf file, the text part is converted into coloring information instead of an independent text part.
So the embedded depth of the painted part is determined based on the paint logic.

NVNDO, 2023-12-11:
Hey @zhimin-zeng-bambulab . going into objects and selecting a different color for each part works much better than painting and has also resolved the problem. 

I future objects that actually require painting, i will have in mind using a 3mf file.

closed 2023-12-11 by NVNDO