bambulab/BambuStudio #2906: I wish a clean layer before text is printed (especially useful for keytags)

NVNDO, 2023-11-17:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: Intel i7 7700K
RAM: 16GB DDR4 CL14
GPU: Nvidia RTX 2060

### Printer

Bambulabs P1S

### How to reproduce

Slice any keytag and scroll down the layers until reaching the base surface

Keytag in Prusa Slicer  (Which also has the automatic color-change function) :
[image]

Completely even surface with straight lines:
[image]

### Actual results

Result in BBL-Studio (Pre-sliced text on flat surface) :
[image]

### Expected results

A flat surface like here:  

[image]

As when there is no text on that layer, why should there be any trace of it? Does not make any sense.

### Project file & Debug log uploads

[KayTag_Test.zip](https://github.com/bambulab/BambuStudio/files/13385247/KayTag_Test.zip)
[debug_network_Fri_Nov_17_00_28_14.zip](https://github.com/bambulab/BambuStudio/files/13385249/debug_network_Fri_Nov_17_00_28_14.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2023-11-17:
#175

QingZhangBambu, 2023-11-17:
@XunZhangBambu add it later

XunZhangBambu, 2023-11-21:
Do not apply "only one wall on top surface" and you can an even surface
[image]

NVNDO, 2023-11-21:
@XunZhangBambu 
Thank you soooo much for helping out! and please forgive me for opening a bug-report.
I didn't know this option was having this effect. testet everything... except of that. 

So glad this was such an easy fix.

You make my day!

NVNDO, 2023-12-12:
> This 'fix' doesn't work for all instances, only where small islands are created. This is still a missing feature.
> 
> #175

If that's the case - I just reopened the Issue. Can you provide more info?

GuyH77, 2023-12-12:
@NVNDO Sure. I can provide some screenshots now but happy to provide a simple project example if needs be. This is a sign i'm making. The text and logo have small open regions below them despite having 'Only one wall on top surfaces' set to Not apply. You can see that the large flat surface of the sign is interrupted which causes inconsistency in the finish due to the way the surface is created. I've dried monotonic and monotonic line etc. Nothing improves it. We should be able to have a full complete top surface of a sign on which the text and anything else prints. The finish will be so much better. Prusaslicer has the function for this and is still better at this as a result. 

[image]
[image]
Horrible open regions and I would think slower print as a result.
[image]

NVNDO, 2023-12-13:
I'll Test with a simple design tomorrow and see what result I get.

NVNDO, 2023-12-13:
@XunZhangBambu @QingZhangBambu Hey there, we definitely have a problem here!! 
I can not only replicate the above mentioned issue, but I also encountered another one related to this behavior. 

I'm printing 4 Coaster of 95mm in diagonal and 3mm height.
The first 2* layers (printing at 0.2 layer height) print in 2 colors, but from there on its only black.

Still, like you may see in the underneath screenshot we are at layer 3* and Bambu Studio is still slicing the image from layer 1 and 2 instead of making a flat surface (which would be much smoother and overall more time efficient)
[image]

Layer 4* is printing like its supposed to:
[image]

Layer 5* is also fine.

BUT Layer 6* is again printing the Logo of layer 1 and 2 - That does NOT MAKE ANY SENSE: 
[image]

Layer 7* is also messed up!
[image]

From there on, all layer until 13 are like they're supposed to be "Flat"

This time I'm on Version: 1.8.0.62

GuyH77, 2023-12-13:
@NVNDO Thanks for the extensive testing. I will check those settings but I fear all of this is a workaround. If the setting that is already in Prusaslicer was implemented it would allow simple control and could combine well with other functions. I raised this in 2022 and it's still an issue for me. 

My issue ticket:
https://github.com/bambulab/BambuStudio/issues/175

XunZhangBambu, 2023-12-13:
@NVNDO @GuyH77 
Sorry, I have already implemented the “interface_shell” mentioned in [https://github.com/bambulab/BambuStudio/issues/175](url) .However it doesn't improve the quality of these layers and i find these flaws are associated with the handling in perimeter generator.  I will start addressing this issue and it will be resolved in version 1.9

NVNDO, 2023-12-13:
@XunZhangBambu thx a lot for the quick response and looking forward to address this problem.
With best regards

XunZhangBambu, 2024-01-10:
@GuyH77 Could you provide the 3mf files ?

NVNDO, 2024-01-10:
Hey @XunZhangBambu 

I just encountered the same problem with a different design. Here's the 3mf in case that helps :)
[Splitting layers.zip](https://github.com/bambulab/BambuStudio/files/13892055/Splitting.layers.zip)

GuyH77, 2024-01-10:
> @GuyH77 Could you provide the 3mf files ?

Here is an example. The text and shape should print on a nice complete smooth layer, we want to option to avoid the holes below them. It's very easy to create examples like this. As I mention, Prusaslicer can do this well. Thanks for working on this issue.

[Interface_layer_issue_example.zip](https://github.com/bambulab/BambuStudio/files/13895581/Interface_layer_issue_example.zip)

EricDimitri, 2024-03-15:
> @NVNDO @GuyH77 Sorry, I have already implemented the “interface_shell” mentioned in #175 .However it doesn't improve the quality of these layers and i find these flaws are associated with the handling in perimeter generator. I will start addressing this issue and it will be resolved in version 1.9

Do we have any idea when 1.9 will be released? at least beta version? Thanks!!!

GuyH77, 2024-03-25:
@XunZhangBambu This still isn't improved/fixed in 1.9 beta. I can't believe this is so complex when it works perfectly in Prusaslicer

3 colour sign. Base in pink with two colours for text and border. The pink 'part' should be solid top layers which the border, text and logo print on top. There should be no gaps in the top layers of the pink part, at least this should be an option. 

[image]
[image]
[image]

Is the issue understood correctly? I'm very happy to dedicate time to this if helpful.

Thanks.

XunZhangBambu, 2024-03-26:
@GuyH77 
Still working in progress. We understand  your issue. This is caused by the top_shell_layer. The pink part is top surface and the rest of that layer is detected as sparse infill beacuse its not visible.Use sparse infill there can save filament. In that part above, we use a internal bridge to support the solid layers.

GuyH77, 2024-03-26:
@XunZhangBambu Thanks for the update. It may save a small amount of filament but at the expense of quality. I've wasted more filament than it saves because I have to dispose of the sign. I often have to reprint on a Prusa because I know it will come out well. I want the same experience from my X1C's. Thanks again.

This is similar in PS. Interface shells options means this is treated as a uninterrupted solid layer which is preferred for this application.
[image]
[image]

NVNDO, 2024-04-26:
> @NVNDO @GuyH77 Sorry, I have already implemented the “interface_shell” mentioned in #175 .However it doesn't improve the quality of these layers and i find these flaws are associated with the handling in perimeter generator. I will start addressing this issue and it will be resolved in version 1.9

Well... Just downloaded V1.9 Beta and funny enough, need to print a Label "again", and this problem has still not been resolved.

The Label and Text are two separate objects and should be handled as such, but no,.. 
This is so disappointing.  Half a year later, having invested in 1 P1S, 2 AMS and another 2 A1 Mini and I can still not print Labels as necessary.... My Ender 3 Pro with Cura could do that years ago. 

Sorry... but I'm a little frustrated, sad and tired at this point. 
I know/assume, my/our problem is not a huge priority, but this can't be so difficult.. can it? :/

The sign I needed to print:

[image]
[image]
[image]
[image]

GuyH77, 2024-04-27:
> > @NVNDO @GuyH77 Sorry, I have already implemented the “interface_shell” mentioned in #175 .However it doesn't improve the quality of these layers and i find these flaws are associated with the handling in perimeter generator. I will start addressing this issue and it will be resolved in version 1.9
> 
> 
> 
> Well... Just downloaded V1.9 Beta and funny enough, need to print a Label "again", and this problem has still not been resolved.
> 
> 
> 
> The Label and Text are two separate objects and should be handled as such, but no,.. 
> 
> Half a year later, having invested in 1 P1S, 2 AMS and another 2 A1 Mini and I can still not print Labels as necessary.... 
> 
> My Ender 3 Pro with Cura could do that years ago. I now.. that doesn't help, but its still a fact. 
> 
> 
> 
> Sorry... but I'm a little frustrated, sad and tired at this point. 
> 
> 
> 
> The sign I needed to print:
> 
> 
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [image]
> 
> 
> 
> I wouldn't be so sad if this was the only label I had to print as on some I was able to mitigate this by turning on "Top surfaces" or "Topmost surface" but sometimes its still not possible to get it right.
> 
> 
> 
> And here is the final result: 
> 
> 
> 
> [image]
> 
> [image]
> 
> 
> 
> A totally messed up top surface due to inconsistent lines printed around those letters
> 
> ****

I'm testing the latest 1.9 beta and as you have found it's a complete mess. It's actually regressed and is worse than 1.8

This issue has been present since the first version and I share your frustration. I repeat this often but it's amazing to me that Prusaslicer handles this perfectly and yet Bambu Lab can't implement the same solution when they based this slicer off of it. 

Please BL find a solution to this problem. It's hugely detrimental to the quality of prints from your machines.

GuyH77, 2024-04-27:
[image]

This is from a clean profile, standard settings in a new file.

NVNDO, 2024-04-27:
I'm actually at the point to buy a different printer as I can not wait any longer. I'm about to start a business and have been focusing on different aspects on my current projects to at least get going, but I need this feature to be implemented and working, otherwise I'm out with Bambu - at least in a business-aspect.

My problems will not change a thing, but that's how it is.

XunZhangBambu, 2024-04-27:
> ![image](https://private-user-images.githubusercontent.com/106596720/326174142-b5da5a0f-bfeb-4128-841e-acd8ef6632b7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

Sorry, but in beta version, we have readd the interface shell and use a new infill alogirthm. Open the interface shell option and the problem you mentioned in your previous 3mf was solved. The interface shell option can help the issue if you use multiple color.

[image]

[image]

GuyH77, 2024-04-27:
@NVNDO You'll be able to print that sign perfectly with a Prusa printer and Prusaslicer. Doesn't need to be a Prusa printer but I can recommend the XL. It's been great for me. 

I'm tempted to try make a working profile for the X1C in Prusaslicer. I shouldn't have to but I think it's necessary now.

GuyH77, 2024-04-27:
> > ![image](https://private-user-images.githubusercontent.com/106596720/326174142-b5da5a0f-bfeb-4128-841e-acd8ef6632b7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQyMTcwNjIsIm5iZiI6MTcxNDIxNjc2MiwicGF0aCI6Ii8xMDY1OTY3MjAvMzI2MTc0MTQyLWI1ZGE1YTBmLWJmZWItNDEyOC04NDFlLWFjZDhlZjY2MzJiNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyN1QxMTE5MjJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jZTljNjNiMDJiZWZmMDNjZDdiOTg0NzEzZjRhM2NkMTExMzUzYWJiY2Q2Y2ZjOTZmYjdkNzgyYWNmN2FiNDMxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.-fkuIKPsD26dCvFrQWPCOV1bd91YNe5iI5AslL9iqgs)
> > This is from a clean profile, standard settings in a new file.
> 
> @GuyH77 Sorry, but in beta version, we have readd the interface shell and use a new infill alogirthm. Open the interface shell option and the problem you mentioned in your previous 3mf was solved. The interface shell option can help the issue if you use multiple color.
> 
> ![image](https://private-user-images.githubusercontent.com/139755545/326175198-f62b59d9-0882-461a-8e70-cf608b8366cc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQyMTgwNzksIm5iZiI6MTcxNDIxNzc3OSwicGF0aCI6Ii8xMzk3NTU1NDUvMzI2MTc1MTk4LWY2MmI1OWQ5LTA4ODItNDYxYS04ZTcwLWNmNjA4YjgzNjZjYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyN1QxMTM2MTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iYmNkZmVmNTQ2NGIxMGRmZmJkODEzNGEwOWZiMTUxYjE4NzkzZDE0NjdhMWRkNWIwNDcxZjFiNGU5ZjAyNzdhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.6sWnAhCuhLDFSzQAxryz5c8K7f6XSSX62wJw9BLK51s)
> 
> ![a38ad679-c40c-46e2-a857-5211cb38e11f](https://private-user-images.githubusercontent.com/139755545/326175151-604bd263-750e-4b75-98f9-2e9894870318.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQyMTgwNzksIm5iZiI6MTcxNDIxNzc3OSwicGF0aCI6Ii8xMzk3NTU1NDUvMzI2MTc1MTUxLTYwNGJkMjYzLTc1MGUtNGI3NS05OGY5LTJlOTg5NDg3MDMxOC5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MjdUMTEzNjE5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWQzN2MwNTMzMTQ1ODJjYjQ5OGY1OGVhOTI0NWZjN2Q1YTZkMDVmOWNiYThhYTQ1NzA4ZjcwZGMzMmQxZDdlNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Z2ko9v5jLoItyduH4BzybuPsMzxpG0m_eANknY-kpUg)

That's great! Why can't I see this option in the latest beta? Is this yet to be released? Thanks.

XunZhangBambu, 2024-04-27:
@NVNDO 

> That's great! Why can't I see this option in the latest beta? Is this yet to be released? Thanks.

@GuyH77  That's in develop mode and has been released. 
We split the layer for a better adhesion. Also in some cases, when the lower layer is dark color  and the upper layer is top surface with light color, we will permeate the top surfaceshape downwards to prevent the infiltration of the color from the lower layer.

GuyH77, 2024-04-27:
> @NVNDO
>
>
> @GuyH77 That's in develop mode and has been released. We split the layer for a better adhesion. Also in some cases, when the lower layer is dark color and the upper layer is top surface [...]

@XunZhangBambu Thank you. I didn't realise you'd added it in developer mode. I've just tried it with that test file and it is better but you still interrupt the top layer below the text. It should be one single clean layer to get the print quality required. 

[image]

NVNDO, 2024-04-27:
I'm truly happy for you all @GuyH77 @QingZhangBambu but in my case its still not working. 

I just updatet to 1.9.1 - installed it and restarted the PC

Here in Standard 0.20 settings without changing anything:
[image]
At least it looks better as before (exactly like in Orca Slicer - which I tried this morning)

Switching in "interface shells":
[image]
Doesn't do a thing for me....

Then I changed Only one wall on top surface to - "Not applied" like you @GuyH77 suggested 
[image]
Without any change....

I tried Only one wall on first layer..
[image]

..and changed Classic to Arachne just for the heck of maybe getting a different result...
[image]
but nothing.

Maybe I'm doing something wrong here and I'm just not noticing it,
At this point in time I'm starting to seriously question the reason for my exístanse as whatever I try to do or archive, it just wouldn't work.  I'm giving up.

OzDruiD, 2024-04-27:
@NVNDO I can try your file? Or create my own i guess.  I used the example above, Ziggy... and just changed purely interface shells, resliced and instantly noticed it - screenshot here at layer 18 as example below, but does it throughout.

Before interface shells. - layer 18
[image]

After interface shells - layer 18
[image]

and without that "Only one wall on top suerffaces"...Not applied,  I can confirm i get the same top layer interruption - layer 20 in his example
[image]

"Only one wall on top surfaces" - Not applied - layer 20 again (final top layer before text) - no intteruption.
[image]

NVNDO, 2024-04-27:
EDIT: Or not!

Well... It works - but ONLY with different colors. (No idea why that is, but that's enough for me at this very moment)
Seems like I previously did not color the Text and therefore it did not work. 

Here an overview of how it behaves with and without colored text: 

Everything in (1) color:
[image]
Does NOT work.

In (2) colors:
[image]
Works.

Pffff.... I wish I'd get payed for all these hours.. but well.. Its FINALLY DONE!! THANK YOU!!

I still can't believe it... 

Thank you so much @GuyH77 for being part of this from the get go and helping throughout all this time.
A huge THANK YOU @XunZhangBambu for keeping the work and implementing this just on time. Having this working means the world to me - Thank you from the bottom of my heart. 

Also a big thank you @OzDruiD for helping out on the community forum and especially coming here to help - I really appreciate your time and effort.

Thank you so much everyone!

OzDruiD, 2024-04-27:
I actuallyt just created mine in makerlab, and had similar issues so yeah itts kinda wierd.
Layer 4 - nothing 
[image]
Layer 5 - the prior layer to text and graphics.. soemthing is blending
[image]
Layer 6 - 1st of text top layers
[image]

OzDruiD, 2024-04-27:
@NVNDO  Maybe one colour doesnt work because well, it doesnt really matter, your not going to notice colour bleed or issues with the same colour, I just wonder though if will still cause printing defects like OP posted.

NVNDO, 2024-04-27:
@OzDruiD 

Yeah, something is off when certain colors are selected. 

XunZhangBambu mentioned this:  `Also in some cases, when the lower layer is dark color and the upper layer is top surface with light color, we will permeate the top surfaceshape downwards to prevent the infiltration of the color from the lower layer.` 
which to be honest I did not fully understand as colors could infiltrate whatever color is on top or bottom. 

Also the scenario says: dark color on top - light color at bottom. 
In my case its the other way around and creating problems. Not sure... I think they might need to give this some more attention, but for me it seems good so far. 

`I just wonder though if will still cause printing defects like OP posted.`
I will let you know in two hours - just started the print (but from what the preview shows, it should be perfect)

NVNDO, 2024-04-27:
`I agree, I see in single colour the same issue as you.`

Yep! pretty strange...

OzDruiD, 2024-04-28:
So on the black top layer, same colour , defintely imprinted.  You can see the "Danger" logo here in black
[image]
and .. whilst needs more flush,   it just came out terrible finish as well.
[image]

NVNDO, 2024-04-28:
@OzDruiD  Let's see what XunZhangBambu says about that..

PD: Could you attach your file so I can have a look at it..?

XunZhangBambu, 2024-04-28:
> `I agree, I see in single colour the same issue as you.`
> 
> Yep! pretty strange... but that should not be an issue, because who on earth prints labels in only one color (for sure there will be someone!) XD so yeah.. they might want to have another peak at this.

We will permeate the top surface shape downwards and print it as a solid fill to increase strength.
Let's use a simple model to show.
[image]

[image]

[image]

When you set "top_layer_shell" to 1 and "top_layer_shell_thickness" to one layer height , the top surface will not permeate to lower layer.

XunZhangBambu, 2024-04-28:
@NVNDO  Enable the"interface shell" option ,we will handle areas with different params areas separately ,e.g., areas with different color.That means the top surfaces of different colors will not permeate to the current layer.

OzDruiD, 2024-04-28:
@XunZhangBambu  Thanks - both your top shell and the interface shell, different parameters explains the  2 scenarios we see.
GREATLY apprecaite the effort for the screenshots and info.    

It was as a I suspected for 1 scenario, diferent parameters, eg colour - automatically gets treated differetly. with the "interface shell" option turned on.  

The other scenario, same type\colour etc - still visible - makes sense now and also as suspected and as per the notes - but the screenshots of how to change top shell count and thickness to stop it happenng really helps.

XunZhangBambu, 2024-04-28:
@OzDruiD You're welcome! I'm glad I could help.

cutmoney, 2024-05-02:
I can confirm this finally works for me as well by enabling Develop Mode and enabling interface shells, a LONG awaited feature. It did crash a few times on the Mac version though for some reason.

NVNDO, 2024-05-13:
@XunZhangBambu Hey there, is it just me... or is it again not working on the recent release?
I'm starting to get a headache from this...

I have a sign (again) and nothing is working.
Not activating "Interface shells", nor changing to any of the 2 available options on "Only one wall on top surface".
Changing "Top shell layers" does also not work.

@GuyH77 & @OzDruiD you having any trouble?

XunZhangBambu, 2024-05-14:
> @XunZhangBambu Hey there, is it just me... or is it again not working on the recent release? I'm starting to get a headache from this...
> 
> I have a sign (again) and again its not working. Activating "Interface shells", nor changing to any of the 3 available options on "Only one wall on top surface" works. Changing "Top shell layers" doesn't do anything either...
> 
> Standard 0.20 layer height profile: ![mycreation13 05 2024_20_58_33](https://private-user-images.githubusercontent.com/73851760/330174143-24927822-1f97-4be6-8c59-aba5ac8f2ba2.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2NTU5NTUsIm5iZiI6MTcxNTY1NTY1NSwicGF0aCI6Ii83Mzg1MTc2MC8zMzAxNzQxNDMtMjQ5Mjc4MjItMWY5Ny00YmU2LThjNTktYWJhNWFjOGYyYmEyLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE0VDAzMDA1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY2MDM5Y2ZkMzdiNWVkYjE2ZTk5ZjVhMmFhOTlkNmIyYzI1ZTZkM2ZhMTIwYTc1N2M4MWFkYWNmZWM0NmViODQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.GRN5cLig8KswqUDZNOZlQEN-Ncv2IDqrTol4Xa1UbVo)
> 
> Interface Shells ACTIVATED: ![mycreation13 05 2024_20_57_54](https://private-user-images.githubusercontent.com/73851760/330174228-c34b9377-6d3f-4fb2-8d18-48c589b2051a.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2NTU5NTUsIm5iZiI6MTcxNTY1NTY1NSwicGF0aCI6Ii83Mzg1MTc2MC8zMzAxNzQyMjgtYzM0YjkzNzctNmQzZi00ZmIyLThkMTgtNDhjNTg5YjIwNTFhLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE0VDAzMDA1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJhOTJiYTdlYjY2ZDM4ZmE0YjY1YTllOGM5MmQzMzgyMDNkNDFkNzkyNzZmM2U1OGQ5OTRiYzNjZTAwODM0ODQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.lWjDCHP17I0dcj-jALOzIaYwmsIXzWH9rED65IERl_8)
> 
> * Only one wall on top surface (NOT APPLIED):
>   ![mycreation13 05 2024_20_57_00](https://private-user-images.githubusercontent.com/73851760/330174398-357c8fda-4ee2-4189-94e0-d2dda348f30b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2NTU5NTUsIm5iZiI6MTcxNTY1NTY1NSwicGF0aCI6Ii83Mzg1MTc2MC8zMzAxNzQzOTgtMzU3YzhmZGEtNGVlMi00MTg5LTk0ZTAtZDJkZGEzNDhmMzBiLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE0VDAzMDA1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZiMmZhOWY3ZDhhM2M1MDM3ZjM5NTlkZjgzZDNmYzljNjdjMzY4MmMzODQwYTkyMzM0NjM0YmZjNTg4ODM5N2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.QDhHu2eABiqba4xR-ft4v8bRLhqcI6JL8CnMq8II1Dg)
> 
> What am I doing wrong?
> 
> @GuyH77 & @OzDruiD you having any trouble?
> 
> PD: The only way I get a better result is when applying more "wall loops" but then I get gaps through several layers where no infil no nothing is applied.
> 
> EDIT: I'm back to 1.8.4.51 ![mycreation13 05 2024_21_23_13](https://private-user-images.githubusercontent.com/73851760/330180135-e17bc0b0-4b7e-49e9-90f7-f45b9f94be13.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2NTU5NTUsIm5iZiI6MTcxNTY1NTY1NSwicGF0aCI6Ii83Mzg1MTc2MC8zMzAxODAxMzUtZTE3YmMwYjAtNGI3ZS00OWU5LTkwZjctZjQ1YjlmOTRiZTEzLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE0VDAzMDA1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ4YzMyMTA4ZTY0YzYzZmQyZjdmYzMxODgyYmRhZTA0MzlkMzE4ZWM1NTgyZGVlY2VkYjc1N2ZiYjFjM2IxM2ImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7zQQJbefYoyPctfi_XISt6-VCwMS-8yWyuuud5YDKLw)
> 
> Almost perfect - except this: ![Screenshot (604)](https://private-user-images.githubusercontent.com/73851760/330180515-eec5a3d3-41d2-48dc-bd83-6bed68313ad9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2NTU5NTUsIm5iZiI6MTcxNTY1NTY1NSwicGF0aCI6Ii83Mzg1MTc2MC8zMzAxODA1MTUtZWVjNWEzZDMtNDFkMi00OGRjLWJkODMtNmJlZDY4MzEzYWQ5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE0VDAzMDA1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU2Mjk3NmE0OWE2NTlkNmQxMmUzNGFmOGU1OTcyNzczMGI2YzBlMWM4MGRhMDI3OTU1ZGE0ZjEzMTk3YTExZjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ypZOyZfK2rGIwpuUK-SUiAYznb4PbE7OH5JV3gXrsIc) ..but I will mitigate this with 100% Infill.
> 
> Bambulabs - I'm DONE!

Could you please share the 3mf ?

NVNDO, 2024-05-17:
Hey @XunZhangBambu 

I think we have a mayor flaw here - no wonder some have problems and other don't. 

After several hours of testing (I can really not afford to do this any longer..) I could still not figure out what exactly is causing this to happen. I approached to problem by creating a simple label and altering its shape bit by bit and then adding a part of mine.

See here: 

4 different designs (prime tower disabled) for better view
[image]

Here we see the top-layer inconsistencies:
[image]

Gap fill is created but overrun by the top layer (I had never seen this happen before...)
[image]

Activating "Interface shells" same result.
[image]

FIRST FLAW!!

Previously I had made a screenshot of the exact same settings and the outcome was a different one.  See here:
[image]

Changing "Only one wall on top surface" to- Not applied.
[image]

All looks good right..?!

Well... WATCH THIS: 
[image]

The exact same everything - Just another "it" label added. 

Suddenly One is good - the other one is bad. HOW ON EARTH?!

DeanDavis, 2024-06-03:
I'm getting the same issue.
Mac version 1.9.1.67
Developer mode on.
Interface Shells turned on
Only One Wall on Top surface set to Not Applied
STL and 3MF attached.
Layer before color change
[image]

[Letter O.zip](https://github.com/user-attachments/files/15538097/Letter.O.zip)

DeanDavis, 2024-06-03:
So the issue is that I'm doing a layer color change. My top object, ie. the Text isn't a separate object so the programming logic doesn't see it as a separate "material"/"volume". It should.
I've put in a feature request to PrusaSlicer.
Should I duplicate it on BambuSlicers GitHub?

https://github.com/prusa3d/PrusaSlicer/issues/12776

XunZhangBambu, 2024-06-05:
@DeanDavis  This is an intended behavior

exeljb, 2024-06-08:
In order for my print to not experience those impressions on the top layer of the base platform, I couldn't have the wall loops set to 1 or 2 loops.  Once I set the wall loops to 3 or higher, those impressions go away.  I also have develop mode on, interface shells checked, top shell layers = 1, and 'only one wall on top surfaces' set to 'not applied'.
[image]
[image]

NVNDO, 2024-06-25:
@XunZhangBambu
> Could you please share the 3mf ?

Here is another one. I can unfortunately not share the entire file, nor the other ones.
[image]

The problem persists in most of my signs and KeyTags and I'm seriously mad about this. 
So mad, that I'm back to using my Ender 3 Pro for these things in conjunction with Cura.

I can't understand why its so hard to just get an even layer throughout the entire part (including the top layer) because what is done on purpose here at least with the top layer- does not suite everyones needs, like mine, previous posters, @DeanDavis and many more around the Community forum. 

[Base_A.zip](https://github.com/user-attachments/files/15976109/Base_A.zip)

XunZhangBambu, 2024-06-26:
@NVNDO  If what we done on purpose doesn't suit your needs, you can disable it. Take the Base_A.3mf as an example , set the top_shell_layers to 0 and you will get what you want.
[image]

NVNDO, 2024-06-27:
Hey @XunZhangBambu 
I just started again from scratch and it now worked. 

I'm not sure if I maybe have/had an adjustments made that interfered somehow, but I tried with all my willpower and didn't get it to work. I does now though. 

BUT what if one prints a Label, Keytag or whatever that has Infill and therefore needs a top layer shell? 
If I set top_shell_layers to 0, there is no top layer... What solution is there for that scenario? My Label I print have infill and therefore do nor work with the given solution.

XunZhangBambu, 2024-06-28:
@NVNDO  If you want top layer and doesn't want top layer to affect the layers underneath , set the top shell layer to 1.In summary, you need to understand the function of this parameter, then you can explain the behavior mentioned above.

TinyMito, 2024-07-08:
@NVNDO did you try to increase the top object by 1 wall loop?

[image]

See more tests I have done here: https://forum.bambulab.com/t/solution-raised-text-on-3d-prints-without-top-surface-defects/82927/8?u=kip

NVNDO, 2024-07-08:
@TinyMito Hey Mito, ahm... im not sure. I might have (when desperately trying to get it to work - with maybe a different all-together setting) but I will recheck that when I have time. 

Side-note: Due to mysterious mal-functioning of both my A1 Minis (only when printing those tags) and my P1S I temporarily abandoned all serious projects on my Bambu machines. 

Using my Ender 3 Pro with Cura right now. Slow but (at least) it works.

I'll be back to you (in the Forum) 
Thx a lot for your help. 🙏🏼

DeanDavis, 2024-07-08:
I haven't read all the previous posts but I have been pretty successful in getting smooth layers sliced under my color changed layers.
First, if I'm using an actual multi-color file where each color is it's only STL then checking "Interface Shells" and setting "Only one wall on top surfaces" to "Not Applied" is all I need to do to get smooth laters under a color change.

But if I'm doing a color change at a layer height I need to still check "Interface Shells" and set "Only one wall on top surfaces" to "Not Applied" but also add a "Height Range Modifier".
I set the lower bound the the layer just under the color change and the upper bound to the color change layer.
I then just set "Top Shell Layer" on the modifier to one more than standard
Here is a before and after doing that...

[image]

[image]

NVNDO, 2024-08-26:
@DeanDavis It seems to depend on the design. I sometimes get it to work by just activating the "interference shells", sometimes I have to change from "Only top layer" to "Not applied", Other times none of that works..

Since the newest release I'm having less trouble - not sure if they changed anything else or its just me being lucky on my designs.

rinkek, 2025-01-10:
"Interface shells" works sometimes, and sometimes it doesn't. Playing with the amount of toplayers seems to make a difference, not sure why.

EricDimitri, 2025-01-10:
> "Interface shells" works sometimes, and sometimes it doesn't. Playing with the amount of toplayers seems to make a difference, not sure why. Also where you start your color change. Try to paint as close (down) as you can to the top layer, the color must change immediately after top layer And it only works with a color change, that's probably why you need to change color as close to top layer as possible

Yep. It is full of workarounds all over, but, We need a fix. This is crazy. @XunZhangBambu, Are youworking on this specifically? Do we have a version number, or an ETA of when will this be addressed?

Thanks a lot!!!

rinkek, 2025-01-10:
Imho this is a significant request for people who want to print badges or signs. 
It works, but only after an immediate color change after the top layer. So this doesn't work for people without an AMS.
Like EricDimitri said; allow to set the toplayer to be flat before additional other stuff on top is printed.
I can imagine this is not something you want for other prints, so maybe it could be set on the layer itself? Like you can add a pause or custom gcode?

EricDimitri, 2025-02-28:
@XunZhangBambu Any update on this? It is crucial for many prints... IS it in the roadmap?

thanks!!!

[6 more replies]