bambulab/BambuStudio #2907: 0.2 Nozzle Presets for P1P in .10mm Standard seem off compared to the other presets for 0.2 nozzle

mhwunder, 2023-11-17:
### Bambu Studio Version

1.8.0.57

### Where is the application from?

Bambu Lab github releases

### OS version

Mac  14.1

### Printer

P1P

### How to reproduce

If I compare the built in presets for the 0.2 nozzle with each other, the 0.10mm setting has more differences than the others.

To elaborate:
The only difference I see when comparing presets from the 0.06mm setting to the 0.08, 0.12, and 0.14 is the layer height.  Example of 0.06 to 0.14
[image]

When I compare the other 0.2 nozzle presets to the 0.10mm setting there are additional changes than just the layer height.  I don't think this is correct.
[image]

### Actual results

I manually adjust the 0.10 preset to make the only difference the layer height.  I made a user preset instead of using the built in one.

### Expected results

I would expect the 0.10 preset to only differ from the other 0.2 nozzle presets only by layer height.  Of course, the 0.10mm setting may be the correct one and the others are wrong.

I am not sure.

### Project file & Debug log uploads

No project file needed.  This can be seen with a new project and no model imported. By using the Compare Presets tool.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2023-11-17:
file just changes the setting differently with 0.2.

mhwunder, 2023-11-17:
So is the built in preset for .1 layer height correct?  I’m not sure I follow your statement.

QingZhangBambu, 2023-11-20:
0.06 and 0.14 profile is a quailty version be added recently, which slow down the speed for better quality. 0.1 is the a version for fast print speed which may not good for some small model. I advice you to slow down the acceleration for better quality. the quality preset of o.1 layer height will be added later.

mhwunder, 2023-11-20:
Ok.  I understand now.  Thank you for taking the time to explain.

closed 2023-11-20 by mhwunder