bambulab/BambuStudio #2908: Print Job

CnRcreationstn, 2023-11-17:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Venture 13.3.1

### Printer

Bambu Lab X1C

### How to reproduce

After a print completes I get a printer busy with another job error when trying to print the next plate. I have to quit out and restart to clear the error

### Actual results

Unable to print new plate until I restart the slicer

### Expected results

Print the next job

### Project file & Debug log uploads

It happens with all my files.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-11-23:
Hi. @CnRcreationstn 
Please use the latest version to confirm this issue again.
https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.62

walterwongbbl, 2023-12-21:
If you have any other questions, please create a new one. The current conversation will be closed.

closed 2023-12-21 by walterwongbbl