bambulab/BambuStudio #2909: 涂色后切片没有颜色，仍显示单色

Hotinosin, 2023-11-17:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 11 23h2

### Additional system information

CPU：AMD 5600
GPU：RX6600

### Printer

P1P+AMS

### How to reproduce

涂色后切片，切片无颜色

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[bambu_scraper_grip_ams.3mf.zip](https://github.com/bambulab/BambuStudio/files/13386873/bambu_scraper_grip_ams.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Hotinosin, 2023-11-17:
上来看，发现1.8.0 Public beta发布了，安装1.8.0后，切片倒是有颜色了

zhimin-zeng-bambulab, 2023-11-24:
这个BUG已经在 1.8.0 Beta 版本修复了

closed 2023-11-24 by zhimin-zeng-bambulab