bambulab/BambuStudio #2911: Unable to remove X1 Carbon printer from Presets

MarkDerbyshire, 2023-11-17:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home, Windows 10 Home

### Printer

Bambu Lab P1S

### How to reproduce

1. Go to Printer
2. Select Printer Dropdown
3. X1 Carbon is still listed even though it is deselected in Printer selection popup
[image]
[image]

### Actual results

Bambu Labs X1 carbon still shows in dropdown

### Expected results

Bambu Lab X1 carbon not to show in dropdown

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-11-21:
Please provide application logs and configuration files (BambuStudio.conf).

How To:
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

nickr1957, 2024-01-06:
Any update? I want to deregister my P1P and reregister it as P1S now I have firmware that can say it has ben upgraded...

bambu123, 2024-01-08:
> Any update? I want to deregister my P1P and reregister it as P1S now I have firmware that can say it has ben upgraded...

We didn't reproduce it. Please provide application logs and configuration files (BambuStudio.conf).

An21V1rus, 2024-06-14:
I am having this exact same problem on version 1.9.2.57.

[image]

[image]

[Config and Logs.zip](https://github.com/user-attachments/files/15842508/Config.and.Logs.zip)

An21V1rus, 2024-06-15:
I actually think I figured out what is happening for me, as I'm fairly new to Bambu Studio itself. 

When I load a 3MF from MakerWorld, it seems to set the project up using whatever the creator had set for the file. I have to close Bambu Studio and open a blank project to get the X1C printer to disappear. 

Not sure if that's intended functionality, but I could see why it would be.

DanBao-Bambu, 2024-06-19:
May I ask if the opened file contains GCode? If it does, the preset will be displayed, even if it is not checked in the wizard.

An21V1rus, 2024-06-19:
Yes, I believe that is the case. I've been seeing it off and on depending on the randomness of where I'm getting my files (MakerWorld, Thingiverse, etc.)