bambulab/BambuStudio #2912: Permit 'Add Pause' context option on vertical layer selector with 'By Object' print sequence.

JohannesMP, 2023-11-17:
**Is your feature request related to a problem? Please describe.**
When Print sequence is set to 'By layer', in preview mode you are able to right click the vertical layer selector and add a pause.
[image]
This permits you to add embedded items such as magnets or nuts.

However when Print sequence is set to 'By object' this right click menu is entirely greyed out.
[image]

**Describe the solution you'd like**
I would like to be able to use the 'Add Pause' right click option when Print sequence is set to 'By object'. 

I understand that the other options such as changing filament or adding custom g-code may be problematic, but it would be really useful to be able to add pauses even when printing multiple objects.

Context: I am using my AMS units to print multiple models, each with its own color, in 'By object' mode to avoid filament changes, and am embedding magnets in them. I am aware that I would need to wait for each pause to be hit to add the magnet, but this would be worth it for me vs. having to print by layer with multiple filaments and the filament waste that would be associated with that.

victornpb, 2024-05-20:
Apparently the feature was in an incomplete state https://github.com/bambulab/BambuStudio/issues/1596#issuecomment-1522599436. Would be great to have this feature completed. 

Other related issues:
https://github.com/bambulab/BambuStudio/issues/3158
https://github.com/bambulab/BambuStudio/issues/3669
https://github.com/bambulab/BambuStudio/issues/3726

wcaleballen, 2024-06-14:
+1 
I agree, this issue should be addressed.

gleep52, 2024-08-17:
+1 here as well - I'd love to add a pause by object to change filament to print clones in different colors.  It would save the bed leveling and startup time for small prints that I can fit multiple on the bed in object mode without collisions.

flightoffancy, 2024-12-13:
+1
Was just trying to do this exact thing and discovered it isn't possible on a plate when print by object is set.

dgiaffe, 2025-01-09:
+1 here, just received my printer and would love to see this so I can switch color between objects since I don't have an AMS system.

leebobleebob, 2025-01-19:
Please make this happen it would be so useful!

lgarron, 2026-01-03:
Cross-linking the other times this has been filed:

- https://github.com/bambulab/BambuStudio/issues/3158
- https://github.com/bambulab/BambuStudio/issues/3726
- https://github.com/bambulab/BambuStudio/issues/4449
- https://github.com/bambulab/BambuStudio/issues/5957

It feels like a rather silly restriction not to be able to pause between objects.

It's certainly possible to start a new print but:

- Sometimes I don't know if I'll be in the room when a given object finishes printing. I would like the print to pause and notify me so I can take it out if I'm around, or just keep going if I'm not.
- Starting a new print takes overhead (slicing, prepping, printer start-up and calibration/leveling, filament retraction and re-loading) that might not be warranted.