bambulab/BambuStudio #2917: cannot load file

nilton61, 2023-11-18:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Additional system information

Intel(R) Xeon(R) CPU E3-1505M v5 @ 2.80GHz   2.81 GHz
16,0 GB

### Printer

banbulab x1c

### How to reproduce

import/import step/select file

### Actual results

Program hangs for over 15 min. Had to terminate
Since program needed to be terminated no project or log file can be exported.

The step file is quite complex but it should not take this long to load, I rebooted computer and tried again with similar results

### Expected results

File should load after reasonable time

### Project file & Debug log uploads

[empty.zip](https://github.com/bambulab/BambuStudio/files/13399610/empty.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/13399615/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

nilton61, 2023-11-18:
I managed to load the model with an stl file, but i really want to use step files

SaltWei, 2024-02-06:
Slicer is using occt lib to import step file. Will try to fix, but lib issue, is not easy to be fixed.