bambulab/BambuStudio #2922: FW: Pressure Advance Doesn't Behave as Expected

CCS86, 2023-11-19:
### Bambu Studio Version

Firmware Related

### Where is the application from?

Bambu Lab github releases

### OS version

Win 10 x64

### Printer

X1C

### How to reproduce

In doing some manual pressure advance calibration, using the scripts built into Orca Slicer, I found some strange behavior.

While the test starts out as you would expect, with some under-extrusion at the beginning of the fast print move, and over-extrusion at the end (not enough PA); as you move through "ideal" PA and into "too much" PA, the behavior is not good.

Material: Bambu PLA
Speed: 20 - 180 mm/s
Accel: 8000 mm/s2
Layer height: 0.2 mm
Line width: 0.6 mm

### Actual results

Rather than seeing the over-extrusion and under-extrusion switch positions on the print line, we see a return to the "under PA" behavior, but worse.

To me, it is looking like the extruder is skipping steps:

[image]

[image]

### Expected results

We should see a trend towards over-extrusion at the fast line start and under-extrusion at the fast line end.

### Project file & Debug log uploads

[PA-lines_PLA_9m49s.zip](https://github.com/bambulab/BambuStudio/files/13404350/PA-lines_PLA_9m49s.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

sanguinaria-canadensis, 2023-11-20:
the low speed region is too slow ( 20 mm/s ), which made the non-linearity of the filament becoming the major issue . 
try enhancing the low speed to 50mm/s .

CCS86, 2023-11-20:
> the low speed region is too slow ( 20 mm/s ), which made the non-linearity of the filament becoming the major issue . try enhancing the low speed to 50mm/s .

I don't agree.

Very often print moves start from a zero head velocity.

CCS86, 2023-12-14:
@QingZhangBambu @sodachen22 

Were you able to reproduce?

CCS86, 2024-01-05:
Any updates @QingZhangBambu @sodachen22 ?

SaltWei, 2024-01-18:
Mon-linearity of the filament issue is not easy to be solved by control algorithm at the moment.
I will ask the printer team for any progress.

CCS86, 2024-02-14:
> Non-linearity of the filament issue is not easy to be solved by control algorithm at the moment. I will ask the printer team for any progress.

Thank you.

I really think my results show skipped extruder steps though. If we focus on the beginning of the fast segment, at very low pressure advance values, we see under-extrusion, as we expect.  

As pressure advance is increased, we start to see a reduction in the under-extrusion. From here, the expected behavior is to slowly transition to a blob of over-extrusion, as pressure advance values become too high. Instead, we see a regression back to under-extrusion. 

If you can verify with software, or a stepper analyzer, that pressure advance is functioning as intended:
[image]

The only explanation for this would be from the extruder skipping steps.

sanguinaria-canadensis, 2024-02-18:
the picture are all seems okay for me , except the top two lines .
the problem is in the gcode there is no "M400" before "M900" cmd , that made the M900 is excecute before the motion is applied .
to solve this , the gcode should change to :
"
...
M400
M900 K[k] L1000 M10
...
"

CCS86, 2024-02-18:
> the picture are all seems okay for me , except the top two lines .  "

I disagree. This is what a proper PA test should look like.

Starting from no/low PA: 

- the acceleration zone should be under-extrusion trending towards over-extrusion.
- the deceleration zone should be over-extrusion trending towards under-extrusion.

[image]

CCS86, 2024-02-18:
This page has a good description:

https://ellis3dp.com/Print-Tuning-Guide/articles/pressure_linear_advance/introduction.html

CCS86, 2024-02-18:
I just repeated my test, but failed to replicate the result.

Maybe something was changed in one of the recent firmware updates? Maybe it was in the g-code (I generated new gcode with Orca).

These results look much better:

[image]

SaltWei, 2024-02-19:
@CCS86 I have sent this ticket to printer firmware team. But maybe you can not get full support of firmware on studio github issue column.

CCS86, 2024-02-19:
I made a video talking through an idea I have to improve PA performance.

https://youtu.be/zZjimJ6Gp7w

Many of the biggest print quality issues these days (square corners, balancing between rounded off and bulged; perimeter seams; etc) consist mainly of slight extrusion errors, with nozzle velocity (starting or ending) near zero. 

If 2 more parameters could be added to the M900 command, to tune the timing of the accel and decel phases of PA application, there is a good chance of improving print quality in a meaningful way.

CCS86, 2024-02-22:
@sodachen22 @SaltWei 

Could my results and video be forwarded to the firmware team for review?

SaltWei, 2024-02-23:
> @sodachen22 @SaltWei
> 
> Could my results and video be forwarded to the firmware team for review?

Got it. Thank you very much!

CCS86, 2024-06-22:
Hi @SaltWei any updates on this issue?

I am calibrating pressure advance for some Bambu PETG and finding the same exact issue, with the latest firmware.

gaolegao-lx, 2024-06-24:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

CCS86, 2024-06-25:
> **Please upload log through Handy APP, and give us your ticket number**
> 

@gaolegao-lx 

US240625551001

Thank you for looking into this.

CCS86, 2024-07-06:
@gaolegao-lx 

Were you able to advance this issue?

CCS86, 2024-07-28:
@gaolegao-lx  @SaltWei  @sodachen22 

Have you guys looked into this? Can you reproduce?

CCS86, 2024-08-28:
@gaolegao-lx @SaltWei @sodachen22

Hello?

CCS86, 2024-12-30:
@gaolegao-lx @SaltWei @sodachen22

Guys, why won't you reply on this issue? It has been 6 months since you requested the support ticket submission.

gaolegao-lx, 2024-12-31:
@CCS86 Hello, I checked into the situation, and we have already considered your suggestion.
But there are many coupled factors in this issue, and we cannot simply add parameters to make the test line look good, as it may have unnecessary impacts and risks on other users and scenarios.

CCS86, 2025-02-12:
> [@CCS86](https://github.com/CCS86) Hello, I checked into the situation, and we have already considered your suggestion. But there are many coupled factors in this issue, and we cannot simply add parameters to make the test line look good, as it may have unnecessary impacts and risks on other users and scenarios. However, thank you very much for your suggestion! We will make efforts to optimize it in the future.

@gaolegao-lx why would you be afraid to test improvements? The PA line test displays the most basic extrusion dynamic control: the ability to print a constant width line, as the head speed rapidly changes. If this is demonstrably not working well, how could improving this possibly reduce print quality?