bambulab/BambuStudio #2923: Add Speed-Modes to Slicer-Speed settings.

NVNDO, 2023-11-19:
**Is your feature request related to a problem? Please describe.**
Its not a problem, rather something pretty unusual and annoying. 
Why do I have to change Speeds AFTER sending the job off to print? That makes no sense.

**Describe the solution you'd like**
I'd like (and apparently many others as well) to set a given Speed-mode before sending the job off to print. 
Please add the Speed-modes with a Checkbox to Speed settings: 

Silent [  ]
Standard [  ]
Sport [  ]
Ludicrous [  ] 

**Describe alternatives you've considered**
NO alternatives. That's the way it should be. 

**Additional context**
[image]

bolsoncerrado, 2023-11-23:
I second this. This could also be implemented in the PRINT JOB tab like this:

[image]

batsboss, 2023-11-26:
I agree with this! too !!!

NVNDO, 2023-11-28:
> I second this. This could also be implemented in the PRINT JOB tab like this:
> 
> <img alt="speed" width="480" src="https://private-user-images.githubusercontent.com/10992914/285124547-ca9bab62-f0e3-4123-8530-57fab821a5f6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDExNjYwNTIsIm5iZiI6MTcwMTE2NTc1MiwicGF0aCI6Ii8xMDk5MjkxNC8yODUxMjQ1NDctY2E5YmFiNjItZjBlMy00MTIzLTg1MzAtNTdmYWI4MjFhNWY2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI4VDEwMDIzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRkYjQ2ZWE3OGY4NDRlNWNmMjQwZTE1ZmRlOThjZjgzMzBkNTdlOWQxMDdiNTAxNTM4NmU4NTRhZDhjY2JkNGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.po2E8L72nRSDSLDHrDvCwmNdLPz8tQn_sXZ9eIfaYaU">

I actually like your idea! That might actually be better. Nice :D

batsboss, 2023-11-28:
Very good

flurl, 2023-12-02:
I'd prefer that setting to be added to the slicing profiles. That way it can be saved and does not have to be adjusted every time one starts a print job.

colton123e, 2023-12-10:
I prefer the silent mode but find it inconvenient to wait to change the mode once the print actually starts. Adding it to the slicing profiles gets my vote so you can set everything up, slice the object and send it over the printer. Plus the slice preview could reflect the time to print for the desired mode.

bolsoncerrado, 2024-01-09:
In fact there's room for the two petitions, meaning, you could always save your desired speed setting for a job and if in a rush, have the ultimate decision on upload time and update the speed accordingly when sending the job idk lol

whatever, please DO any or both!
haha

Xaeco, 2024-02-13:
Be great to have it as an option in the Preview screen so you can assign it at certain layer. This would allow adjusting the speed during delicate parts.

bolsoncerrado, 2024-02-14:
@xaeco that's what heigh modifier was meant for. It's already there.

Xaeco, 2024-02-14:
Yes. Except not all settings are available in height modifier and you need to calculate appropriate values.