bambulab/BambuStudio #2925: Tree support failure + random square trees

Toetta, 2023-11-20:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows10 + + Windows 11

### Additional system information

Tried on multiple PC, same issue

### Printer

X1C

### How to reproduce

Select tree support
Generate
(It wont make tree supports) By switching to slim / hybrid one gets strange square trees)

### Actual results

It generate standard supports

### Expected results

Tree support with round branches.

### Project file & Debug log uploads

[4.zip](https://github.com/bambulab/BambuStudio/files/13415519/4.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Toetta, 2023-11-24:
> It's not a bug. When there are too many tree nodes, the generation of round tree support can be very slow. As a workaround, we make square branches which consist of much fewer vertices and faster to be generated. We'll remove this walkaround once we solve the "generation too slow" issue.

Thanks for your input.I have also issues with very simple parts choosing tree support. It generates normal support unless you set a non default tree style like strong / slim / hydrid..Could this be fixed?Best regardsMalkusSkickat från min iPhone24 nov.

tangketan, 2023-12-21:
@Toetta The default tree style is just hybrid. There should be no problem. Would you please try the latest v1.8 release? If there is still a problem, please share your 3mf for us to diagnose.

vanowm, 2025-02-02:
Perhaps instead of automatically switching, allow user to switch between square or round. I'd rather wait a few extra seconds than dealing with this.