bambulab/BambuStudio #2928: Slicer doesn't slice text

Geekay1972, 2023-11-20:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 14.0

### Printer

P1S

### How to reproduce

1. Load the attached STL file in bamboo studio.
2. In the thickness steps every step as a 0.2 mm high text describing the thickness: 0.2, 0.4, 0,6 ....
3. When you slice the model, the text is not in the generated G-code
4. This happens no matter what layer hight you choose, even with a 0.1 layer its lost
[Geeetech_PLA_Silk_gold.stl.zip](https://github.com/bambulab/BambuStudio/files/13418811/Geeetech_PLA_Silk_gold.stl.zip)

### Actual results

[image]
[image]

### Expected results

numbers should bed printed

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13418935/log.zip)
[Geeetech_PLA_Silk_gold.3mf.zip](https://github.com/bambulab/BambuStudio/files/13418938/Geeetech_PLA_Silk_gold.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

GuzziRaz, 2023-11-21:
There's no bug here. The text is simply too small for your settings (or nozzle). Just change to Arachne (or a 0.2 nozzle) and it no longer disappears.

Geekay1972, 2023-11-22:
Thanks for the answer. Actually I had the very low hight on my list but not the size :-)
Wouldn't it be a great improvement for the slicer to spit some warning in such a case instead of silently dropping the whole text? Could you make a feature request out of this BR?

Geekay1972, 2023-11-22:
Yes, sure. I mean in this case it's not just a detail but the whole text as an object that is below resolution.
BTW, I also liked your hint to the Arachne mode (could also go into the warning). I would have never found out that by myself ;-)

GuzziRaz, 2023-11-22:
Arachne was the default for a while but they reverted that due to some problems (not sure what or how often it occurred - I use it a lot and never had a problem). It will likely be the default again once the bugs are ironed out.

Geekay1972, 2023-11-29:
> Arachne was the default for a while but they reverted that due to some problems (not sure what or how often it occurred - I use it a lot and never had a problem). It will likely be the default again once the bugs are ironed out.

Thank you for this background info. I meanwhile switched over to Arachne and like it much more, seams to produce less gap infill so I was already wondering why its not the default ....

walterwongbbl, 2023-11-30:
@zhimin-zeng-bambulab