bambulab/BambuStudio #2930: CLI generated gcode.3mf not imported correctly (sliced)

wizrads, 2023-11-21:
### Bambu Studio Version

1.8.0.57

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

CPU: 12th gen intel
RAM: 32 GB
GPU: Nvidia 3080 Ti

### Printer

X1C

### How to reproduce

1. Generate any 3mf from 3mf or stl file using the CLI
2. import into bambu studio as a project file

### Actual results

Needs to be re-sliced, even though the g code is in the 3mf file

### Expected results

It should be imported already sliced, as what happens when importing gcode.3mf files that were generated by selecting "Export all sliced file"

I assume what happens is that the import can not read the gcode.3mf created by the CLI. Attached is the gcode.3mf generated by the CLI, and the gcode.3mf generated by saving the project prior to slicing.

### Project file & Debug log uploads

[BothGcode3mfFiles.zip](https://github.com/bambulab/BambuStudio/files/13422472/BothGcode3mfFiles.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

lanewei120, 2023-11-22:
you can add this option "--min-save"
then the generated .3mf only includes .gcodes and the gcodes can be parsed directly in bambustudio

wizrads, 2023-11-24:
Thank you! This is exactly what I needed.

closed 2023-11-24 by wizrads