bambulab/BambuStudio #2931: Studio window pops up when printer sequence changes.

Qjeezy, 2023-11-21:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 pro

### Additional system information

CPU: intel(R) Core(TM) i3-4030U CPU @ 1.90GHz

ram: 16gb

### Printer

Bambu lab P1P 

### How to reproduce

Open 2 instances of bambu studio. 

start a print with one of them. 

then switch to the other one and start working on another project. 

### Actual results

Whenever the printer performing the job from the first window of bambu studio makes a sequence change (ams filament swap, done printing, etc..) the window for that print job will pop up and interrupt the work being performed on the second open window of bambu studio. 

### Expected results

The first instance of bambu studio should update in the background and not open and interrupt other work being performed. 

### Project file & Debug log uploads

Any file will work. I do not have logs. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-11-24:
Hi. @Qjeezy 
Our data push is synchronized with the printer, so if you start receiving print data now, messages will be sent to the Bambu Studio instance regardless of its location.
You can log out of your account in Bambu Studio so that you won't receive printer push notifications.

closed 2023-11-24 by walterwongbbl

Qjeezy, 2023-11-24:
> Hi. @Qjeezy Our data push is synchronized with the printer, so if you start receiving print data now, messages will be sent to the Bambu Studio instance regardless of its location. You can log out of your account in Bambu Studio so that you won't receive printer push notifications.

I think you misunderstood the issue. If I sign out of bambu studio, how will I continue to work on another project in studio within the second instance that is open? Also, if I sign out of studio, how will I monitor the current print status? 

What I tried to explain is that when working within the second instance of bambu studio, when the first instance receives and update it will pop up and take over the second one and interrupt the current work being performed. There are multiple people who dislike this bug. It is annoying and updates to the screen should not cause it to pop up and take precedence over what other work is being performed. It should update in the background.