bambulab/BambuStudio #2938: DEFAULT SUPPORT SETTINGS FOR SUPPORT/RAFT BASE AND SUPPORT/RAFT INTERFACE DO NOT CONSISTENTLY FOLLOW SPECIFIC LAYER  COLORS, COLORS

LROME66, 2023-11-22:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab github releases

### OS version

WINDOWS 10 PRO 64

### Additional system information

Intel(R) Xeon(R) W-11855M CPU @ 3.20GHz   3.19 GHz
64.0 GB (63.7 GB usable)
64-bit operating system, x64-based processor

### Printer

BAMBU LAB X1 CARBON SERIES

### How to reproduce

Set Filament for Support settings to Default

### Actual results

The support(s) are extruded on a  vertically split 67mm tall hollow cone with a 20mm White circumferential start section, 2mm Red circumferential stripe, 16mm White circumferential section, , followed by a 2mm Blue circumferential stripe finishing the remaining 27mm White.  The White section begins with white supports, and extrude Red for the red stripe, the blue stripe extrudes solid, however the support/raft base layers alternate White and Blue .

### Expected results

Blue layers should inclusively with the support and support interface extrude the same color filament
[image]

### Project file & Debug log uploads

[ring finger splint - ring.zip](https://github.com/bambulab/BambuStudio/files/13443920/ring.finger.splint.-.ring.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2023-11-24:
I don't understand your question. The 3mf file seems completely OK for me. The support base/interface will just use the last used filament if you didn't set any specific filaments for them.
Can you upload a big picture with an arrow showing the part where you think is wrong?

LROME66, 2023-11-24:
[image: Picture2.jpg]
 3.jpg
<https://drive.google.com/file/d/1EVmL1imMx0t4wfJncPZ6thoUDkgvrK7B/view?usp=drive_web>

The filament for support should have been blue for the entire element, not
alternate blue and white. The red element filament is used for the support
and element. Settings are default for both support filaments.  Why did the
red filament carry through the element and the support, and the blue
element did not?

(see attached .jpg file 3.jpg.)

Larry RomeStant

      M: 818.231.3014
    http://specialks.net

closed 2024-02-05 by LROME66