bambulab/BambuStudio #2939: Allow for manual entry of K factor on Flow Dynamics page 1.8.0

capsel22, 2023-11-22:
**Is your feature request related to a problem? Please describe.**
I've got a bunch of filament already PA calibrated on another Bambu account. Now, I wanted to move those calibrations to another account, however the "Calibration" page does NOT have "Manage Results" button (since I never ran a Dynamics calibration on that account) and there is no way to just enter the K factor manually without running a calibration. I know I can export filament settings, but it does not include the "Manage Results" entry.

**Describe the solution you'd like**
When creating "Custom filament" allow for entry of K factor and/or when going to "Flow Dynamics" allow for a manual entry and skip the process.

**Describe alternatives you've considered**
Using Orca to save filament with PA value embedded

DokuKaefer, 2023-11-23:
In my opinion, it would also be better if the K value could be entered in the filament profile.
Just as it is possible with the Orca Slicer. Then the value would also be transferable.

lanewei120, 2023-11-24:
the calibration result is not bound to account but machine

after you change your account, is the same printer connected and shown on the "Device" tab?
could you give some picture and upload the logs for our debug?

thanks

capsel22, 2023-11-24:
Hi,
No it's different printer. I've got two P1S combos connected to two different bambu accounts. Just wanted to move filament settings (easy) and calibration settings (impossible?) from one P1S on accounts A to second P1S on account B

zhimin-zeng-bambulab, 2023-11-24:
@capsel22 
The k value is not only related to the material, but also to the machine. The k value of one machine is not suitable for another machine, so the k value is stored on the machine.

pgifford, 2025-04-27:
> The k value is not only related to the material, but also to the machine. The k value of one machine is not suitable for another machine, so the k value is stored on the machine.

The A1 Mini holds only 16 K values.  I have far more than 16 different filaments.  Bambu Lab store has twice that many colors of PLA Basic.  

When the machine cannot store any more results what are we to do?