bambulab/BambuStudio #2940: Auto-Arrange not fitting all item onto plate when enough space is easily available and spacing set to 0

NXSmiggy, 2023-11-22:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

P1s

### How to reproduce

Load project into slicer
Select plate 2 
Hit auto arrange

### Actual results

Items scatter with massive gaps between items, forcing overspill into empty space
[image]
[image]

### Expected results

Items should be more arranged as below, plate has plenty of empty volume
[image]

### Project file & Debug log uploads

[Spacing issue.zip](https://github.com/bambulab/BambuStudio/files/13444436/Spacing.issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

GuzziRaz, 2023-11-22:
FWIW if you set distance to non-zero (such as 0.1 or 0.01), it works just fine. It's the zero distance that bugs out.

tangketan, 2023-11-24:
When setting distance=0 (which means "auto distance") and enabling tree supports, objects are separated by a relatively large gap. The reason is, tree supports can have a large span which we don't know exact value before slicing, and to avoid possible collision we have to use a maximum distance. There are two solutions:
1. set distance to a non-zero value just as @GuzziRaz mentioned.
2. disable support

GuzziRaz, 2023-11-25:
> When setting distance=0 (which means "auto distance") and enabling tree supports, objects are separated by a relatively large gap. The reason is, tree supports can have a large span which we don't know exact value before slicing, and to avoid possible collision we have to use a maximum distance.

Ahaa that's good to know. Please convert this issue to a feature request for having this information in a tooltip!