bambulab/BambuStudio #2942: Unable to restart connection after standby on MacOS

parkerbol, 2023-11-23:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 14.0 23A344

### Additional system information

M2 PRO
16Go

### Printer

Bambu A1 / Bambu P1P

### How to reproduce

1. Open Bambu studio
2. Connect to the account
3. Connect to printers (2 printers)
4. Wait for standby for example (10 min)
5. Mouse move
6. Navigate Bambu studio still opened
7. Refresh printers by selecting a printer ...
8. Unable to refresh (cf capture)
[image]

### Actual results

Need to disconnect and reconnect

### Expected results

able to reconnect to cloud services

### Project file & Debug log uploads

[debug_network_Mon_Nov_20_21_01_22.log.zip](https://github.com/bambulab/BambuStudio/files/13449849/debug_network_Mon_Nov_20_21_01_22.log.zip)
[debug_network_Mon_Nov_20_21_01_22 2.log.zip](https://github.com/bambulab/BambuStudio/files/13449850/debug_network_Mon_Nov_20_21_01_22.2.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

parkerbol, 2023-11-23:
Same bug on another brand new mac with M1 and Macos 1

walterwongbbl, 2023-12-07:
@parkerbol 
Please use the latest version
https://github.com/bambulab/BambuStudio/releases/tag/v01.08.01.57

walterwongbbl, 2023-12-07:
Hi. @parkerbol 
cloud_conn_callback::connection_lost!,
I have found many lost connections in the log you provided. May I ask if your network is stable? I suggest using a wired connection or not too far away from WiFi. Please confirm if this will be effective.

walterwongbbl, 2024-03-11:
This issue will be closed

closed 2024-03-11 by walterwongbbl