bambulab/BambuStudio #2944: Having trouble sending file to A1 printer - not my X1 or my P1P - just the A1

SylvesterMoltisanti, 2023-11-23:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

11th Gen Intel(R) Core(TM) i9-11900K @ 3.50GHz   3.40 GHz - ram - 64.0 GB (63.6 GB usable)

### Printer

Bambu Lab A1

### How to reproduce

slice
print plate
send
[debug_network_Thu_Nov_23_07_56_59.log](https://github.com/bambulab/BambuStudio/files/13450894/debug_network_Thu_Nov_23_07_56_59.log)

### Actual results

It runs through the process of uploading, usually goes to 100% but then says failed with connection

### Expected results

It should load and start purging leveling and running

### Project file & Debug log uploads

[debug_network_Thu_Nov_23_07_56_59.log](https://github.com/bambulab/BambuStudio/files/13450906/debug_network_Thu_Nov_23_07_56_59.log)
[debug_network_Wed_Nov_22_13_05_01.log](https://github.com/bambulab/BambuStudio/files/13450909/debug_network_Wed_Nov_22_13_05_01.log)
[debug_network_Sun_Nov_19_10_43_20.log](https://github.com/bambulab/BambuStudio/files/13450915/debug_network_Sun_Nov_19_10_43_20.log)
[debug_network_Sun_Nov_19_07_44_11.log](https://github.com/bambulab/BambuStudio/files/13450918/debug_network_Sun_Nov_19_07_44_11.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2023-11-24:
Hi. @SylvesterMoltisanti 
Please use new version.
https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.62

walterwongbbl, 2023-12-04:
Hi. @SylvesterMoltisanti 
Furthermore, I have found in the logs that your requests frequently time out. Have you placed the printer in a location far from the Wi-Fi signal?

SylvesterMoltisanti, 2023-12-04:
Hello.
Thank you for reaching out and offering the insight. You are right, once I
updated to the newest version the problem went away and it has not been an
issue at all.
Not sure why it was timing out. That did happen, but it went through the
same process and was complete then timed out until the newest version was
installed.

Have a great day.

Kind regards,
Sal Moltisanti