bambulab/BambuStudio #2945: Supports working bad on new beta versions 1.8+

dalet11, 2023-11-23:
### Bambu Studio Version

1.8+

### Where is the application from?

Bambu Lab github releases

### OS version

Win 11

### Printer

X1C

### How to reproduce

Im using PLA support for the interface, and the behavior of the support is very different and kinda results in a failure.

See the picture below, the top one is printed with a stable version, second with beta. support interface was impossible to remove, and also looked weird..
[image]

**_see how it looks on the slicer:_**
NEW 1.8+ line 26+30:
[image]
[image]

**OLD stable  line 26+30:**
[image]
[image]

It's missing some interface layer between the support and the support PLA material, they also look out of alignment at the new version.

### Actual results

very bad support, causing failure of the print.

### Expected results

I expect the supports to look like on the stable version, or better.

### Project file & Debug log uploads

[support issues.zip](https://github.com/bambulab/BambuStudio/files/13452433/support.issues.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2023-11-24:
Yes, it's a bug. Will fix it soon. Thanks for reporting.

tangketan, 2023-12-21:
Offical 1.8 has been released and this bug is fixed. Close this issue.

closed 2023-12-21 by tangketan