bambulab/BambuStudio #2946: Setup with custom filiments Blocker

masc2279, 2023-11-23:
### Bambu Studio Version

1.8.0.62

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

Threadripper 64 Core / 256GB of Ram / Nvidia 4090

### Printer

Bambu X1 Carbon

### How to reproduce

Setup up Custom Filiments, Remove/Clear all Systrem Filiments since they are no longer needed since you have custom filiments. And click confirm. You are not allowed to continue since the System thinks you do not have any filiments and states you have to select one filiment must be selected. Well you have one or many custom filiments and its clearly not looking at those and blocking your progression.

### Actual results

The system blocks you from moving forward requiring you to add an un needed filiment or you cannot leave the setting screen for filaments

### Expected results

Actual Path should not only look at system filaments but custom and if there is one filiment no matter system or custom you should not be blocked from continuing. A custom filiment should not be locked to only be able to be used when a system filiment is present. Its not a second class filiment it should be considered a filament period.

### Project file & Debug log uploads

This is not needed. This is an application UI / Requirments issue

### Checklist of files to include

- [ ] Log file
- [X] Project file

DanBao-Bambu, 2023-11-24:
Hello, we have carefully considered your suggestion, but based on the usability of the software, we have protected it during design without system presets.

masc2279, 2023-11-24:
The way you wrote that you are saying this is a feature ? This is a bug, So your saying your not going to fix this bug ? Why is it a good thing to make users scroll thru junk to get to their custom filiments when that is the only thing they use ? And there is no way to get rid of the junk filiments you are forcing them to be there.  This is just bad UI and a bad flow.