bambulab/BambuStudio #2947: Height range modifier in objects

UVM24, 2023-11-23:
### Bambu Studio Version

v1-7-7-89

### Where is the application from?

Bambu Lab Official website

### OS version

Win10

### Additional system information

Bambu Lab Studio

### Printer

P1S

### How to reproduce

Height range modifier in Objects
[image]

### Actual results

Only one range is possible. There is no + or - visible

### Expected results

further rages should be possible

### Project file & Debug log uploads

like that
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

LiZ-Li-BBL, 2023-11-24:
Hello,
This issue has been fixed. We recommend you update BambuStudio to the latest version https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.62

LiZ-Li-BBL, 2023-11-24:
Hello,
You should find the "+" and "-" button in the previous level
[image]

UVM24, 2023-11-24:
Hello,

 

Sorry, there is no + or - button visible.

I checked the program for the latest version. It is the latest version.

I read in the forums that the + on d - can be seen in the English language setting > Negativ.

I checked all program settings, deleted the program and reinstalled it, there is no + or – visible

 

 <https://user-images.githubusercontent.com/120457312/285334310-5c542ccc-2abe-4d57-9d49-e3cc02b66b2e.png>

LiZ-Li-BBL, 2023-11-24:
Hello,
You can use the latest beta version here. https://github.com/bambulab/BambuStudio/releases/tag/v01.08.00.62
And an official version will be released soon.

UVM24, 2023-11-24:
Hello,

Where can I find the beta version?

UVM24, 2023-11-24:
Hello.

 

I found the version and installed it.

Unfortunately no + or - visible!

 

​

UVM24, 2023-11-24:
Hello,

 

I found the beta version and installed it.

I previously uninstalled the old version with a special program. To be sure that everything old is deleted

Unfortunately there is still no + or - visible!

LiZ-Li-BBL, 2023-11-27:
> Hello, You should find the "+" and "-" button in the previous level [...]

Hello,
Can you take another screenshot of this pane? And make sure that you have selected the management view of multiple ranges: "Layers" instead of a single range.

UVM24, 2023-11-27:
Hello,

For me there is only one range possible. But I would like to create several ranges. Like the one circled in my picture above «?».

I can't add any more ranges in my program.

 

 

I found a picture on an American internet site that shows that it was possible :

 

 

I found this on GitHub

 

Please also read:

https://github.com/bambulab/BambuStudio/issues/1715

UVM24, 2023-11-27:
Hello 

 

Addendum. 

Also possible in prusa.

LiZ-Li-BBL, 2023-11-28:
Hello,
Did you post any pictures? Maybe the upload failed and we can't see them.

UVM24, 2023-11-28:
Hello Liz

 

I received a video from BBL.

So my question was answered.

Thank you very much for your help.

Because the video is too big I can't forward it.

LiZ-Li-BBL, 2023-11-29:
That's really nice. Thanks for your reply.

closed 2023-11-29 by LiZ-Li-BBL