bambulab/BambuStudio #2950: IRONING DIRECTION / ANGLE

andrei030700, 2023-11-24:
**Is your feature request related to a problem? Please describe.**
Ironing direction is not present int the Bambu slicer software. as us can see in the screenshot 
[image]
Also this feature is presented in the Bambu lab wiki as u can see 
[image]

**Describe the solution you'd like**
To implement back this feature in the slicer 
**Describe alternatives you've considered**
to be able to set my ironing direction. 

Thank you!

QingZhangBambu, 2023-11-27:
[image]
open develop mode then you can see ironing direction

andrei030700, 2023-11-27:
Hi
I did check the developer mode on, and I saw other functions appearing, but not the IRONING DIRECTION.
Is there something else that i can do?
[image]

SaltWei, 2023-11-28:
> Is there something else that i can do?

Use the latest version on github. It's now included in beta version

closed 2023-11-28 by andrei030700