bambulab/BambuStudio #2952: An option to disable automatic switching to "Device" tab after print is started

azaz44, 2023-11-24:
**Is your feature request related to a problem? Please describe.**
I don't like that Bambu Studio switches to "Device" tab automatically when starting the print. It's good for many poeple but not all. I would prefer to do this manually when needed.
Switching to "Device" affects performance of other applications on PC, especially games.

Plus, some of us prefer monitoring with a smartphone.

**Describe the solution you'd like**
Can we have an option in Bambu Studio preferences:
- automatically switch to "Device" tab after starting a print [ YES / NO ]

So we could disable automatic switching to "Device" tab.

**Describe alternatives you've considered**
The only alternative now is
- wait until job is transferred to the printer (can take long time) and it switches to "Device"
- manually switch back to "Preview" tab

Which is what I mostly do.

ayhanyildiz, 2025-01-02:
This is really annoying since "camera" window is preventing PC to go to sleep mode. Please make this happen.

blorenz, 2026-11-02:
Make this happen. This is a small and simple change that affords a better UX to the more advanced users.