bambulab/BambuStudio #2953: setup wizard is blank

Vance1122, 2023-11-24:
### Bambu Studio Version

1.8.0 beta2

### Where is the application from?

Bambu Lab github releases

### OS version

Mint based on Ubuntu 22

### Printer

X1C

### How to reproduce

Start it up and go to add printer.

### Actual results

Wizard screen comes up and it is blank.

### Expected results

Wizard should start up with text and options I can see and select.

### Project file & Debug log uploads

[debug_Fri_Nov_24_11_49_51_8203.log.0.zip
[debug_network_Fri_Nov_24_11_55_17.log.enc.zip](https://github.com/bambulab/BambuStudio/files/13461960/debug_network_Fri_Nov_24_11_55_17.log.enc.zip)
](https://github.com/bambulab/BambuStudio/files/13461957/debug_Fri_Nov_24_11_49_51_8203.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2023-11-27:
Please refer and try the solutions here #1828.

Vance1122, 2023-11-27:
I think my issue relates to a security certificate as I have to confirm I want to use it every time I start the program. I will take a screen shot of it later. I did somehow manage to set up the X1C by blindly clicking in various spots on the blank set up wizard. So I can use it for my X1C at least, but if I want to set up a different printer, the wizard is still blank and I can't tell what I am selecting.

bambu123, 2023-12-20:
Does this method solve the problem?

WEBKIT_DISABLE_COMPOSITING_MODE=1

Vance1122, 2023-12-20:
Do not know how to use that method.  I just download the appimage and then
set it to open like a program in properties.

bambu123, 2023-12-21:
> Do not know how to use that method. I just download the appimage and then set it to open like a program in properties.
> […](#)

Here is the method
https://github.com/bambulab/BambuStudio/issues/1828#issuecomment-1768736346

Vance1122, 2023-12-21:
OK. It works when I use the WEBKIT line. But I can only run it from the
command line and it needs to be run like that everytime. So my desktop
shortcut does not work.

bambu123, 2023-12-21:
We'll add this to code.

AtomicRPM, 2023-12-25:
> @Vance1122 Does this method solve the problem?
> 
> WEBKIT_DISABLE_COMPOSITING_MODE=1

That did it for me!  Thanks!  Source compiled on Gentoo.

closed 2023-12-25 by bambu123