bambulab/BambuStudio #2954: Error Filling X1C

gvogele, 2023-11-24:
### Bambu Studio Version

Last

### Where is the application from?

Bambu Lab Official website

### OS version

W11

### Printer

X1C

### How to reproduce

the Print fills Gaps with Fillings. There is no Filling to do.

### Actual results

Part not usable

### Expected results

No Fillings

### Project file & Debug log uploads

[shower adapter.zip](https://github.com/bambulab/BambuStudio/files/13462227/shower.adapter.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

gvogele, 2023-11-24:
With Tree Mode it works, ??

gvogele, 2023-11-27:
Yes Correct it shows everyting correct but if i Print it will make fills

QingZhangBambu, 2024-01-17:
could you upload your gcode.3mf? The concerned staff tends to reproudce it.