bambulab/BambuStudio #2956: MACOS: can't input chinese to "3D Text"

wang-yupu, 2023-11-25:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 14.0 public beta

### Additional system information

Apple M1
8GB RAM

### Printer

P1S

### How to reproduce

[image]
在输入文本那栏里无法使用自带输入法输入中文或其他需要使用输入法输入的语言，无法复制粘贴

### Actual results

[image]
在输入文本那栏里无法使用自带输入法输入中文或其他需要使用输入法输入的语言，无法复制粘贴

### Expected results

[image]
在输入文本那栏里无法使用自带输入法输入中文或其他需要使用输入法输入的语言，无法复制粘贴

### Project file & Debug log uploads

没有

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-11-27:
@wang-yupu 想和您这边再确认一下两点
1、您其切换其他下载的输入法（如搜狗），是否能够正常的在 Studio 上输入
2、您用自带的输入法，能够在其他新安装的软件环境下正常输入吗

wang-yupu, 2023-12-02:
对于问题一:
1. 切换至搜狗输入法无法在Studio上输入中文
2. 系统自带的手写也输入不了中文
对于问题二(系统自带输入法)：
1. Orcaslicer同样无法输入中文 
2. 其他应用可以正常输入中午

zhimin-zeng-bambulab, 2023-12-19:
这是我们使用的三方库 imgui 的BUG（mac 系统上无法输入中文），我们暂时无法解决这个问题，抱歉

broven, 2025-03-08:
请问这个issue现在状态是什么？ 我在mac上尝试输入中文，是口口口 不确定是字体的问题还是 这个bug引起的

wang-yupu, 2025-03-08:
最新的Public Release，系统14.2.1，复制粘贴输入可行，但是系统自带输入法按中英键是切不到中文的

x2x4com, 2025-05-18:
我从其他地方复制进来的中文是方块，字体从黑体到SimSong都是

x2x4com, 2025-05-18:
[image]