bambulab/BambuStudio #2958: Height Range Modifier constantly crashing (Winx64)

RipperDrone, 2023-11-25:
### Bambu Studio Version

1.8.0 Beta

### Where is the application from?

Bambu Lab github releases

### OS version

Win11 x64

### Additional system information

core i7, AMD6800XT

### Printer

BambuLab X1C

### How to reproduce

Load model, add height range modifier -> crashing

### Actual results

5secs hourglass, then crashing

### Expected results

height range added (used to work before)

### Project file & Debug log uploads

[Rear Cap 1-piece.zip](https://github.com/bambulab/BambuStudio/files/13463551/Rear.Cap.1-piece.zip)
[debug_network_Sat_Nov_25_10_42_24.log.zip](https://github.com/bambulab/BambuStudio/files/13463556/debug_network_Sat_Nov_25_10_42_24.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

LiZ-Li-BBL, 2023-11-27:
Hello,
Is there any crash.log file or .dmp file in the log file directory? Please provide more information to help us resolve the issue.

RipperDrone, 2023-11-27:
All logs in the directory have been uploaded in my previous post as a .zip file. There's nothing else I could provide 🤔

LiZ-Li-BBL, 2023-12-06:
Hi,
Does this crash occur consistently? Please confirm this directory after the crash occurs again.
[image]

RipperDrone, 2024-10-09:
hasn't crashed on me since..

closed 2024-10-09 by RipperDrone