bambulab/BambuStudio #2963: Studio Crash During Camera View via Arc A770 Graphics

jtbrough, 2023-11-26:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Microsoft Windows 11 Pro 10.0.22631 Build 22631

### Additional system information

CPU: 13th Gen Intel(R) Core(TM) i5-13600K
RAM: Installed Physical Memory 64.0 GB
Integrated Graphics: Intel(R) UHD Graphics 770
Discrete Graphics: Intel(R) Arc(TM) A770 Graphics
Intel ARC driver: 31.0.101.4972 (latest)

### Printer

Bambu Lab X1 Carbon (01.06.05.01) + AMS (00.00.60.40)

### How to reproduce

1. Can recreate with or without print active
2. Navigate to Device tab
3. Select Status page
4. Select Play button to enable X1C camera

### Actual results

100% crash every time if graphics are being displayed via the discrete Intel Arc A770 GPU.

Note: If I swap my display to use the integrated Intel UHD 770 the camera display within Bambu Studio works fine. It also works to display the camera from Bambu Handy on iOS.

### Expected results

The camera view should work using any modern graphics system.

### Project file & Debug log uploads

[bambu-studio.exe.17104.discrete-crash.zip](https://github.com/bambulab/BambuStudio/files/13465605/bambu-studio.exe.17104.discrete-crash.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2023-11-27:
Please refer https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view
Problem 5 - The program hung when loading the video stream! (Windows Only)

jtbrough, 2023-11-27:
Thank you. In this case NVIDIA control panel isn't relevant since none of their products are involved.
Intel suggests that this can be done using Graphics Settings in Windows 11 System Settings ([reference](https://www.intel.com/content/www/us/en/support/articles/000090168/graphics.html)):

[image]

[image]

Unfortunately, now Bambu Studio simply hangs indefinitely instead of crashing immediately when attempting to preview device video after having set graphics to `Power saving` which uses the integrated graphics.

bambu123, 2023-11-27:
It crashed in the graphics driver, we need some time to analyze it. First, we need to help provide the file for this dll.

C:\Windows\System32\DriverStore\FileRepository\iigd_dch_d.inf_amd64_3c4b0f9f7b8d6d5a\igd9trinity64.dll

[image]

jtbrough, 2023-11-27:
[2963-igd9trinity64.zip](https://github.com/bambulab/BambuStudio/files/13470002/2963-igd9trinity64.zip)

bambu123, 2023-11-27:
> [2963-igd9trinity64.zip](https://github.com/bambulab/BambuStudio/files/13470002/2963-igd9trinity64.zip)

Thank you, I still need d3d9.dll.
C:\Windows\System32\d3d9.dll

[image]

AndrewC772, 2023-12-13:
Hi, I have the same issue but i have no integrated graphics as I'm using an AMD 3900x and my gpu is the intel Arc A770 as well. I only included most recent logs as the size was too large to upload and will include the requested dll as well. 
[log.zip](https://github.com/bambulab/BambuStudio/files/13654745/log.zip)

[dll.zip](https://github.com/bambulab/BambuStudio/files/13654747/dll.zip)

bambu123, 2023-12-13:
> Hi, I have the same issue but i have no integrated graphics as I'm using an AMD 3900x and my gpu is the intel Arc A770 as well. I only included most recent logs as the size was too large to upload [...]

Can Windows Media Player in your system play video files?
[image]

AndrewC772, 2023-12-13:
yes
[image]

AndrewC772, 2023-12-13:
yes, It can play videos
[image]

atran315, 2023-12-13:
I have the same problem, using Intel i9-14900k and Intel Arc A770. It just crashes every time. OBS Studio is working, even though it is not convenient.

warthog9, 2024-01-04:
I'll chime in, on 1.8.2.56 with an AMD 7900, which does have an integrated gpu, and an Intel ARC A770, same issue.

I'm guessing that since DX9 is being emulated entirely for the ARC series drivers there's something slightly wonky going on there, so it could be an issue that needs to be brought up with Intel's Arc team (they should be pretty approachable about this).

Setting Bambu Lab to actually run on the Intel Arc only: crashes
Switching it to run on the AMD (though the monitors are only attached to the Arc): video never starts, but clicking anywhere results in a window that's clearly trying desperately to crash (greyed out, spinning blue thing, etc).

If there was a way to pull up the video feed in VLC I'd be more than content to just ignore the video from within Bambu Studio

bambu123, 2024-01-04:
> I'll chime in, on 1.8.2.56 with an AMD 7900, which does have an integrated gpu, and an Intel ARC A770, same issue.
>
> I'm guessing that since DX9 is being emulated entirely for the ARC series drivers [...]

If you are using X1 printer, you can start Go Live and use VLC to connect it.
https://wiki.bambulab.com/en/software/bambu-studio/virtual-camera

warthog9, 2024-01-18:
the VLC work around works, though obviously not ideal.  Will confirm the crash is still present (as I think is expected) in 1.8.4.51

ciaagent, 2024-02-10:
Same issue here. Using Intel Arc A770. No integrated graphics (Ryzen 7 2700X). Although it used to work fine, then one day stopped.

AndrewC772, 2024-03-18:
Appears to be working for me now on Intel ARC driver 5333 hadn't tried it in a while, tried again today and it appears to be working, not sure if anything else has contributed to it. I am on 1.8.4.51 of Bambu Studio.

warthog9, 2024-03-18:
Actually based on @AndrewC772 's comment, I happened to have a print open and gave it a whirl with an A770 and also driver 5333, and it loads up and doesn't crash.  The video is *CHOPPY*, and not anywhere near as smooth as what's on Android, but that's a rather big improvement

ciaagent, 2024-03-18:
Also based on the notifications. I went and checked my intel version and it was 31.0.101.5194, checked Bambu (1.8.4.51) again and it crashed. I updated my intel drivers to 31.0.101.5333 and as @warthog9 stated. It no longer crashes, but unusable levels of choppiness. I can also confirm that it runs smooth on Android. It also used to run smooth in Bambu Studio on windows until it started crashing.

Laess3r, 2024-06-21:
>  I updated my intel drivers to 31.0.101.5333 and as @warthog9 stated.

I can confirm that an update of my ARC drivers fixed at least the app crash! I don't see any choppyness at all :-)

fahaaaxiki, 2024-12-06:
Hi， we have solved this issue in our latest Public version：[1.10.1 Public Release](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50), you can update to this version to check if it works