bambulab/BambuStudio #2967: Duplicate Variable Layer Height between objects

drpeppershaker, 2023-11-26:
**Is your feature request related to a problem? Please describe.**
Not a problem necessarily, but it would be an excellent feature. Currently you cannot use a prime tower when printing multiple objects with variable layer heights. The solution, or workaround, is to select all the objects on the plate and assemble them and apply the _same_ variable layer height to all of them. 

This works okay, except now you have one giant assembly--which renders the 'skip objects' feature of the printers useless. Because if all of your objects are one assembly you can't skip a single one any more.

**Describe the solution you'd like**
I would like to be able to apply the same variable layer height settings to multiple objects without needing to assemble them together. Similar to how you are able to copy & paste Height Range modifiers.

**Describe alternatives you've considered**
An alternative solution would be to manually go in and create a height range modifier that matches the layer height on a per layer basis and use that on each object. However, this is very difficult to do on models with many layer height changes.

**Additional context**

[image]

shiinee, 2024-01-17:
+1. I'd like to be able to have both the prime tower and a purge object(s) on my multicolor prints with variable layer height. It would be fine with me if the variable layer height calculation either adapted to the purge object's shape, or just copied the layer height settings from the primary object. Currently no such print configuration is possible even with the assembly workaround, because you can't designate one part of an assembly as a purge object.

lordbbk, 2024-03-12:
+1. The only useful way of utilizing purge to object, if you duplicate the main object, and use it as a purge target, ignoring the colors... 

Just simply allowing copying the layer heights to another object would be a simple solution - since it will be just an initial state, you can reset it via the standard way, using the variable layer height menu.

Bathsheba, 2026-03-22:
+1
I have multiple objects that are the same height and need the same Variable Layer Height profile. I don't want to merge them because then I can't skip a single failed object, and also they have many parts and it would ruin my organization.

A different feature which would also solve this problem is, a way to group objects without merging them, and Variable Layer Height can be applied to the group.