bambulab/BambuStudio #2968: Stop using non-inclusive terms

reedy, 2023-11-26:
**Is your feature request related to a problem? Please describe.**
Programming has historically used many terms that are non-inclusive. In all cases, there is no need to continue using them, and better alternatives are availble

**Describe the solution you'd like**
Replace "blacklist" (and "whitelist"), at least in the UI with inclusive terms such as:

>denylist, deny, exclude, prohibit, forbid, mute, block, ban

There are other words that should also be replaced, but I've not noticed any specific examples, yet.

**Describe alternatives you've considered**

**Additional context**
* https://www.zdnet.com/article/uk-ncsc-to-stop-using-whitelist-and-blacklist-due-to-racial-stereotyping/
* https://help.apple.com/applestyleguide/#/apsg1a3a0436?sub=apd9aaca4623
* https://source.android.com/docs/setup/contribute/respectful-code
* https://www.ncbi.nlm.nih.gov/pmc/articles/PMC6148600/
* https://docs.microsoft.com/en-us/style-guide/bias-free-communication
* https://www.vice.com/en_us/article/8x7akv/masterslave-terminology-was-removed-from-python-programming-language

djengis, 2023-12-31:
only a small but loud minority things this word is offensive. please ignore.

(wait till he heres about the master-branch)

reedy, 2024-01-01:
It's a good job that GitHub didn't change the default branch away from master. Oh, wait. They did.